pibooth/pibooth #340: help !! with canon CP1300

XoVooth, 2022-05-02:
Hello everyone,

I am looking for someone who uses a canon selphy Cp1300 with the Pibooth!

because I'm going to explode!!!!!!!!

I can not even say what I did, see what I did and look for a lot of config on the net.

in any case on the first use, everything was ok, in terms of printing because I had not seen that I was on wifi on the printer.

but I don't want to be on wifi but on usb!

and since then it does not print

it stays in the spool but no print.

short, I'm looking for someone who could ask me questions from the beginning, because I really lost the way with the config of the CP1300

Thank you for your help

anxuae, 2022-05-02:
Hi @XoVooth,

Several issue are about CP1300, maybe you can find some helpful info:

- https://github.com/pibooth/pibooth/issues/314
- https://github.com/pibooth/pibooth/issues/268

I've never personally configure such a printer.

XoVooth, 2022-05-02:
thank @anxuae 
 I had already read and done it 314

but I had not seen the 268

I will see that 

thank you!

XoVooth, 2022-05-02:
Well! I don't know if I'm progressing!
but after reading the last post
I realized that printing a text via the mousepad works very well with the printer!

so I am at the stage that when using the pibooth, the photos remain in the pool, then an error appears to tell me that I must check "active" in the printer config.

which of course I have already done.

I keep looking and I tell you more!

anxuae, 2022-05-04:
HI @XoVooth, I've seen that you have put a comment in https://github.com/reuterbal/photobooth/issues/38,

If you have a step-by-step procedure to configure the CP1300, can you write it here?

Thanks

XoVooth, 2022-05-05:
Hello @anxuae 
yes it works, I'm still at the photobooth assembly stage.

I'm looking to do a procedure and post it here.

closed 2022-06-15 by anxuae