pibooth/pibooth #344: Dark image

otalchriis, 2022-05-05:
I have a problem!

The images captured by the camera are coming out dark, strange is that in the preview when the timer shows the image is good, only later, both in the original photos folder and in the final file the photo is very dark.

I am using a Logitech C920 webcam.

fab1o978, 2022-05-09:
Did you try to change ISO values for preview/capture? Not sure they have any effect with a webcam though

otalchriis, 2022-05-09:
I tried to change the ISO value, it had no effect. I also tried to install some drivers but also notI have another webcam but it is much inferior, strange is that this problem does not happen to her.
Enviado do Yahoo Mail no Android

anxuae, 2022-08-04:
Hi @otalchriis did you solved your issue?

anxuae, 2023-03-02:
No more info since long time

closed 2023-03-02 by anxuae