pibooth/pibooth #347: Have picture half width, then duplicate it to have duplicate pictures printed

0xecute, 2022-05-13:
Hi,
Thanks again for the project.

I'm trying to find a way to print twice the same picture on a photo. The idea is that people will cut it manually and will both of them get a physical picture.

For example, I have a 4x6 paper, and I want an intermediate output of the picture of 2x6 size, and then it is clone and merge to print one picture of 4x6 and cut it manually.

Is it already possible?
If not, can you point me in the right direction to do it with a plugin for example?

Thank you in advance,
Have a nice week end

0xecute, 2022-05-13:
I just realize the issue is raised here already:
https://github.com/pibooth/pibooth-picture-template/issues/8

Looks like there is no easy solution at the moment :(

werdeil, 2022-05-13:
Hi,

On the contrary the answer on the topic you provide shall fit your need (at least it was intended to):
>if you want to duplicate your photo, you can use the option in the config file:
```
# Print 1, 2, 3 or 4 picture copies per page
pictures_per_page = 2
```

0xecute, 2022-05-13:
No it doesn't, as I want this only when there is three pictures, and when there is only one then it doesn't duplicate :)

werdeil, 2022-05-13:
Ok then I'll update your title as it is misleading

0xecute, 2022-05-13:
Sure.
I finally found a solution.
I edited "pibooth-picture-template" plugin so I can have multiple times the same pictures inside a template, same for the text.
This is a more flexible solution, but maybe overkill if we just want 2 picture_per_page.

Thanks again for this great project !
Have a nice week end

werdeil, 2022-05-13:
Nice, if possible open a pull request with your changes, it will help others

0xecute, 2022-05-13:
I will. I saw it was an issue for others too.

0xecute, 2022-05-14:
Done:
[https://github.com/pibooth/pibooth-picture-template/pull/11](https://github.com/pibooth/pibooth-picture-template/pull/11)

anxuae, 2022-08-04:
[pibooth-picture-template](https://github.com/pibooth/pibooth-picture-template) v1.1.0 released

closed 2022-08-04 by anxuae