pibooth/pibooth #348: Run plugins with root/sudo

emilmusik, 2022-05-15:
**Is your feature request related to a problem? Please describe.**
I am trying to get pibooth to work with adafruit neopixel library but i have a problem. The neopixel library opens from the pibooth plugin file but requires sudo/root to work correctly.

**Describe the solution you'd like**
I would like the ability to have pibooth open plugins with root/sudo

**Describe alternatives you've considered**
Maybe there is another solution to this, maybe another neopixel library or a workaround for the sudo thing.

Thanks.

anxuae, 2022-06-13:
Hi @emilmusik 

You have to start pibooth using sudo command. What kind of procedure did you test?

anxuae, 2022-08-04:
Hi @emilmusik 

Running a program with root privileges is never a good option. To access to the RPi interfaces without `sudo`, you need to add the `pi` user in the correct linux groups. See the solution here: https://github.com/jgarff/rpi_ws281x/issues/326

closed 2022-08-04 by anxuae