pibooth/pibooth #353: pibooth not running with EOS 250d

lilabeste, 2022-05-27:
Hi guys,

since days im trying to get my photobooth running. 

setup:
Pi 4, 4GB Ram
EOS 250D, AF, Scene Mode, no flash
Pi Touchscreen

Often pibooth wont start because there is no camera detected even though its on and connected. Rebooting the pi, turning on and off the camera and/or reconnect the cable will fix it kinda randomly. Since i dont have a pi cam 2 yet i use the EOS 250D alone. Preview works, photos mostly dont work "oops, something went wring". Im using the standard config. Especially more than 1 Picture seems to cause Problems. piiboth.loh is attached - what am i doing wrong? The Camera is pretty new, checked different cables... Am i missing an important setting für DSLR Only or the EOS 250D?
[pibooth.log](https://github.com/pibooth/pibooth/files/8786374/pibooth.log)

werdeil, 2022-05-29:
Hi @lilabeste, from what we see in the logs pibooth doesn't manage to get a capture from the DSLR. Can you run the ‘pibooth-diag’ command and post the result here ?
https://pibooth.readthedocs.io/en/latest/sources/scripts.html#errors-diagnosis

lilabeste, 2022-05-29:
pygame-menu 4.0.7
Pibooth version installed: 2.0.5
Installed plugins: extra-lights-1.0.1
Listing all connected DSLR camera
No compatible DSLR camera detected

The camera is connected, i tried different cables which are known good ones. The Camera itself states "busy" when connected and sometimes works but its kinda random... 

I then tried an old nikon d90, thats working everytime but crashes when taing 4 pictures:
Pibooth version installed: 2.0.5
Installed plugins: extra-lights-1.0.1
Listing all connected DSLR camera
00 : addr-> usb:001,004  name-> Nikon DSC D90 (PTP mode)

================================================================================
Stating diagnostic of connected DSLR camera
Gphoto2: b'gp_libusb1_open [libusb1.c:437]': b"'libusb_claim_interface (port->pl->dh, port->settings.usb.interface)' failed: Resource busy (-6)"
Gphoto2: b'gp_port_set_error [gphoto2-port.c:1190]': b'Could not claim interface 0 (Success). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.'

The nikon is on JPG Fine , no extra raw. Standard config on the pibooth. I just cant get it working ...

werdeil, 2022-05-30:
About the EOS 250d is photo can't find the device there is not much we could do on our side. However from what I see [here](http://www.gphoto.org/proj/libgphoto2/support.php) the camera is supported.

About the Nikon issue: often the issue about the 4 photos comes from the fact that there is not enough space on the camera to store 4 photos, can you confirm there is a SD card in it?

lilabeste, 2022-05-31:
I got the 250d runnnig - It was a stupid mistake. The default standby time was 10! seconds. So when i started the Pi and Camera together the cam was in standby before Pibooth was started.... 

The Problem with Nikon D90 is still freeznig with "smile" on the preview when making nmiore than 1 pic. I checked the SD card in the camera and there is plenty of space. Anything i can try?

anxuae, 2022-06-13:
@lilabeste please check that your DSLR camera is in Manual mode

anxuae, 2022-08-04:
Please re-open if issue is not solved

closed 2022-08-04 by anxuae