pibooth/pibooth #357: RPi 4 & Logi C920

mweedon, 2022-06-10:
On a fresh install (including gphoto2 and opencv) when taking the photo I see a stuttering video, but when the pic is snapped I get 'Ooops something went wrong'.

Seems that it all goes wrong with 'Can not get camera preview image'. Log included...

**Log**
```
[ DEBUG   ] pibooth           : took 0.008 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='photobooth', release='5.15.32-v7l+', version='#1538 SMP Thu Mar 31 19:39:41 BST 2022', machine='armv7l', processor=''
***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.040 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] pibooth           : took 11.237 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ DEBUG   ] pibooth           : took 4.008 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Can not get camera preview image
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 98, in set_state
    hook(cfg=self.cfg, app=self.app, win=self.win)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 69, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 155, in preview
    self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 101, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ DEBUG   ] pibooth           : took 0.003 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='photobooth', release='5.15.32-v7l+', version='#1538 SMP Thu Mar 31 19:39:41 BST 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.027 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] pibooth           : took 3.819 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ DEBUG   ] pibooth           : took 4.015 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Can not get camera preview image
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
[... 190 lines cut ...]

# Print 1, 2, 3 or 4 picture copies per page
pictures_per_page = 1

[CONTROLS]
# How long to press a single hardware button in seconds
debounce_delay = 0.3

# How long to press multiple hardware buttons in seconds
multi_press_delay = 0.5

# Physical GPIO IN pin to take a picture
picture_btn_pin = 11

# Physical GPIO OUT pin to light a LED when picture button is pressed
picture_led_pin = 7

# Physical GPIO IN pin to print a picture
print_btn_pin = 13

# Physical GPIO OUT pin to light a LED when print button is pressed
print_led_pin = 15
```

Any help appreciated!!

mweedon, 2022-06-10:
Solved by reverting to Buster!

closed 2022-06-10 by mweedon

mtz1734, 2023-08-28:
Thanks for the solution

mtz1734, 2023-08-28:
I have the same config (RPi 4 + Logitech C920).

Is it also the case with you that after the preview (after the countdown) the camera takes a long time to take the picture (~ 2-3 seconds)? A new connection to the camera is established, which takes a lot of time.

Is that also a problem for you and do you have a solution?