pibooth/pibooth #358: Sometimes no Photos with Nikon D300s

AndyKausG, 2022-06-11:
Hi There,

on my last Event I had to take the Camera out and be the Photobox myself. :(
Sometimes it works perfekt but sometimes it get stuck. I run Diagnostics an here is the Log.

Additional Informations: 

- If I directly run gphoto2 and take a capture it always works. Even if pibooth-diagnostics shows this error!

Thank you!

[diagnostics.log](https://github.com/pibooth/pibooth/files/8883266/diagnostics.log)

AndyKausG, 2022-06-11:
pi@raspberrypi:~ $ DISPLAY=:0 pibooth 
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: qrcode-1.0.2, picture-template-1.0.0
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
[ INFO    ] pibooth           : Connected to printer 'Canon-SELPHY-CP1300-HTTP'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ ERROR   ] pibooth           : Unsupported option actions/viewfinder=1 ([-1] Unspecified error), configure your DSLR manually
[ INFO    ] pibooth           : Take a capture

anxuae, 2022-06-13:
Hi @AndyKausG, please be sure that your DSLR is in Manual mode (no auto focus)

anxuae, 2022-08-04:
Please re-open if issue is not solved

closed 2022-08-04 by anxuae