pibooth/pibooth #359: Print to different sizes

BramWerbrouck, 2022-06-15:
Hi,

I want to use the DPN QW410 as photobooth printer but before I buy such a printer,  I have a question about the printing. I my case, I let the user choose between 1 photo or 3 photos, when they choose "Print" what will happen? Will the format change automatically?

Let me try to explain this better:
I would use paper 10x15. DPN says it is possible to print three times 5x10 out of paper 10x15 and they are cut automatically in strokes. But how will the printer know if he has to print "1 bigger photo" (1 time 10x15)or "3 smaller strokes" (3 times 5x10) depending on what the user has chosen (1 or 3 photos)?

The constructor of the printer says that it is possible to print strokes of 5x10 out of the paper of 10x15 (so you get the 3 strokes) but this has to be sent to the printer as 1 printjob..

Or maybe there is a other/better solution for a printer? What do you use?

With kind regards
Bram Werbrouck

anxuae, 2022-06-15:
Hi @BramWerbrouck,

There is no possibility to change the number of the picture depending on the user choice.

BramWerbrouck, 2022-06-16:
Hi @anxuae,

wouldn't this be nice in the future, so the user can choose to print 1 big photo (10x15) or 3 small strokes (5x10)? Of course this depends on the connected printer. Is it even possible to code this? What I see now is that the picture saved in the folder is always 2400x3600 (depending if it is landscape or portrait) I would think that if you choose the option 3 photo's, the saved picture would be something like 2400x1200..

anxuae, 2022-08-04:
Hi @BramWerbrouck 

Try using the [pibooth-picture-template](https://pypi.org/project/pibooth-picture-template/), you can specify the size of the page in the template.

Please, re-open if you have any issue

closed 2022-08-04 by anxuae