pibooth/pibooth #364: problems with 2.0.5 / how to downgrade?

DennisD88, 2022-06-24:
I have problems with the new version. But how can I downgrade the software from 2.0.5 to 2.0.4?

werdeil, 2022-06-26:
Hi @DennisD88 can you tell us more about the problems you are having?

You can always install a previous version through pip by specifying the version number (`pip3 install -U pibooth==2.0.4`)but it is not a good idea as we are only working on the last one.

DennisD88, 2022-06-30:
I have a new install with version 2.0.5 but my camera image are shaking. But I downgrade to 2.0.4 and the problem is alsoo. See attachment for the problem. 

https://user-images.githubusercontent.com/90935117/176693288-670129c6-7c67-4ded-990f-8cc6c6872d9a.mp4

after the countdown I get a screen with Oops something go wrong.

DennisD88, 2022-07-05:
I try to solve the problem;
- I format the SD-card and install a fresh install
- Try a other webcam
- Try to downgrade to 2.0.4

All this options was not a solution. Do you have?

DennisD88, 2022-07-08:
- [ ] 

- [ ] 

> Hi @DennisD88 can you tell us more about the problems you are having?
> 
> You can always install a previous version through pip by specifying the version number (`pip3 install -U pibooth==2.0.4`)but it is not a good idea as we are only working on the last one.

See my comments

anxuae, 2022-08-04:
Hi @DennisD88 

Did you solve this issue?

Issue seems to coming from the camera, maybe a bad connection (connector is damaged)? Or you are close to a magnetic disturbance?
I think that downgrading to another pibooth version will have no effect for this kind if issue.

anxuae, 2023-03-18:
No more info, please reopen if issue is not solved

closed 2023-03-18 by anxuae