pibooth/pibooth #365: error pibooth

rafetagarci, 2022-06-27:
/tmp/pibooth.log

werdeil, 2022-06-27:
Hi @rafetagarci, we may need more info on your issue, seems that you paste the path of your log instead of the log itself.

rafetagarci, 2022-06-27:
that´s with a webcam. And conect a picam get this 
file "/usr/lib/python3/dist-packages/picamera/camera.py", line 2270, in _set_resolution
raise picameravalue error(
picamera.exc.picameravalueError: invalid resolution requerted: piresolution(width=1934, heigh=2464)

werdeil, 2022-06-29:
Hi @rafetagarci, as stated in the error you are using a resolution not supported by your camera. You need to put one that is compatible with your camera. Depending on the model you may find compatible resolutions here: https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes

anxuae, 2022-08-04:
Hi @rafetagarci 

Add the correct resolution in the config `[CAMERA][resolution]` according to the camera manufacturer . For a picamera, resolutions are listed here: https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes

closed 2022-08-04 by anxuae