pibooth/pibooth #366: Hi @rafetagarci, we may need more info on your issue, seems that you paste the path of your log instead of the log itself.

rafetagarci, 2022-06-27:
Hi @rafetagarci, we may need more info on your issue, seems that you paste the path of your log instead of the log itself.

_Originally posted by @werdeil in https://github.com/pibooth/pibooth/issues/365#issuecomment-1167144340_
that´s with a webcam. And conect a picam get this
file "/usr/lib/python3/dist-packages/picamera/camera.py", line 2270, in _set_resolution
raise picameravalue error(
picamera.exc.picameravalueError: invalid resolution requerted: piresolution(width=1934, heigh=2464

rafetagarci, 2022-06-29:
[image]

werdeil, 2022-06-29:
Closing the issue as it is followed up in #365

closed 2022-06-29 by werdeil