pibooth/pibooth #367: ALSA error: underrun occurred

Cruiser79, 2022-06-28:
If i start my pibooth i got sometimes directly, sometimes after a while the following error which will then spam the log.
```
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
```

```
pygame 2.1.2 (SDL 2.0.14, Python 3.9.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: ledring-1.0.0, ledring_start-1.0.0
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
error: XDG_RUNTIME_DIR not set in the environment.
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video1): can't open camera by index
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (893) open VIDEOIO(V4L2:/dev/video2): can't open camera by index
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8545:(snd_pcm_recover) underrun occurred

``` 

I don't know if this is a bug, so i opened a normal issue.
The 2 plugin, which where loaded, are only plugins on capture states, which connect to the serial port and send actions to it.
I also couldn't find a similiar issue by now. On the internet i only find issues which described this problems with playing audio. But, as far as i know, pibooth didn't play audios? Anyone can help me to get this logentries away?

werdeil, 2022-06-29:
Hi @Cruiser79 Indeed for me ALSA is linked to sound but except if you use our [pibooth-sound-effects](https://github.com/pibooth/pibooth-sound-effects) plugin pibooth is not playing any sound.

My best guess would be that your issue is linked to your plugins that connect to the serial port, can you tell us more about it?

Cruiser79, 2022-06-30:
The plugins only did serial things, which i now changed to socket-things. You can see one here

```
"""Plugin to start the led ring, when entering in the 'preview' state."""

import pibooth
#import serial
from pibooth.utils import LOGGER
from multiprocessing.connection import Client

__version__ = "1.0.0"

@pibooth.hookimpl

def state_preview_enter(cfg):
#def state_wait_exit(cfg):
    LOGGER.info("Start LED ring")
    LOGGER.info(cfg.get('WINDOW', 'preview_delay'))

    address = ('localhost', 3333)
    conn = Client(address)
    conn.send('s'+cfg.get('WINDOW', 'preview_delay'))
    conn.close()

#    ser = serial.Serial('/dev/ttyUSB0', 9600)
#    ser.write(b's'+str.encode(cfg.get('WINDOW', 'preview_delay')))
#    ser.close()
```

But i found out another thing. I read on the pygame documentary
```
pygame.init()
This will attempt to initialize all the pygame modules for you. Not all pygame modules need to be initialized, but this will automatically initialize the ones that do.
```
So i tried to deactivate the music module, after the init in the window.py
```
        pygame.init()
        pygame.mixer.quit()
```
After that the errors in the logfile disappeared.
So, its a bit hacky solution and not practicable if you use a plugin like [pibooth-sound-effects](https://github.com/pibooth/pibooth-sound-effects) but for me it will work this way.

werdeil, 2022-06-30:
Thanks for the investigation, I'll have a look on pygame side to see if I can do something on our side.

Cruiser79, 2022-07-01:
Thanks @werdeil 
SHall i close this issue, or will you leave it opened as a reminder for your investigation?

werdeil, 2022-07-01:
Keep it open, I'll close it when I find time to have a look.

anxuae, 2023-05-06:
This issue was on PiOS Bullseye which got many update since june 2022.
Issue seems related to ALSA audio config https://forums.raspberrypi.com/viewtopic.php?t=333335

```
default-fragments = 5
default-fragment-size-msec = 2
```

Please re-open is issue is still present

closed 2023-05-06 by anxuae