pibooth/pibooth #368: Stuck on "Take a capture". Is there a timeout possible?

Cruiser79, 2022-07-04:
I already posted a comment under the isuue https://github.com/pibooth/pibooth/issues/356 but open now a new issue for a bigger scope.
If the autofocus of my dslr  camera failed, pibooth stucks on the "Take a capture" and will not reach another state till then. I read al lot of times, that the switch to manual for the autofocus can help with this problem. Autofocus is very useful and i will not miss it, if it is not a must.
Is there not the possibility, that pibooth runs in an timeout after some time, if it stucks on the "Take a capture" mode and restarts himself , like it did for the "Ups! And error occured" message? That will be a more convenience way as one can keep the autofocus to on?

anxuae, 2022-08-04:
Hi @Cruiser79 

Thanks for the tricks, it could be appended to the master to avoid deadlock. but I don' know if we can rely on the `wait_for_event` for a kind of DSLR, so it may introduce regression for some camera model.

Cruiser79, 2022-08-04:
@anxuae what do you mean with your text "but I don' know if we can rely on the wait_for_event for a kind of DSLR,"? Do you thing, that not all DSLRs will support this function?
Or do you think, because you wrote also "it may indroduce regression for some camera model"?, that the 100ms is a way back for some models, because it is such a long time?

anxuae, 2022-08-04:
Events are not well implemented on DSLR depending on the model. In such a case, your code will not work.

anxuae, 2023-04-01:
Implemented in v3.0.0 coming soon

PR #439 (2023-04-01): Fix #368: Gphoto2 event on capture to let autofocus do its job
Close #368
commit 88f95b0d (2023-04-01): Fix #368: Gphoto2 event on capture to let autofocus do its job (#439)

* Use gphoto2 event

* Fix unitests

* Update doc

* Catch exception in emitted events

* Change AsyncTask event on exception
closed 2023-04-01 by anxuae