pibooth/pibooth #369: usb ring light ?

tinni123, 2022-07-06:
any way to make a usb ring light work as a flash ? i have the extra lights plugin working but the lights are too bright so wanna change to a usb ring light any ideas cheers tinni

XoVooth, 2022-07-09:
hello I have a ring led light in 220V.
it's on all night long.

but I'm still struggling a bit with the Nikon D90 settings to find the right settings (iso/aperture/time)

XoVooth, 2022-07-09:
[image]

satcomjimmy, 2022-07-16:
My brother used this plugin. I have not tried it myself yet, but I plan to.
https://github.com/pibooth/pibooth-extra-lights

anxuae, 2022-08-04:
Hi @tinni123 

With the extra-lights plugin, you should be able to drive a relay in order to switch on / off your ring light

anxuae, 2022-08-04:
Seems that @Cruiser79 has developed a plugin for led ring with USB, may be he can share its code. See https://github.com/pibooth/pibooth/issues/367

Cruiser79, 2022-08-04:
@anxuae My plugin is not the correct solution for @tinni123 problem.
I developed a plugin, which communicates with an arduino, which hangs on the usb port. On the arduino is a WS2812B led ring connected to control the leds and made a fading out of all leds one after the other.

Benoit1478, 2022-12-02:
> @anxuae My plugin is not the correct solution for @tinni123 problem. I developed a plugin, which communicates with an arduino, which hangs on the usb port. On the arduino is a WS2812B led ring connected to control the leds and made a fading out of all leds one after the other.

Do you have the code to mettro to control this led strip?

Cruiser79, 2022-12-06:
My plugin and code is divided into three parts. One Code is running on the arduino, one "python server" is running on the pibooth machine and then the plugins for the pibooth application. This seperation with the "python server" was nesseccary, because otherwise ever time i connected to the arduino directly, the arduino reboots and the action, he had to too, was not done.
The solution can send codes to the led strip.
"f" to light all leds
"sX" with an integer as X, to tell, in which time, the led ring shall faded out. So s1 will say, in 1 seconds all leds are faded out on after the other. X is getting from the preview_delay in the config file.

In the meantime, i don't know, why i don't connected the led strips directly to the raspberry. I think this could be also a valid solution. But my solution is now working: never change a running system ;-)

So, here are my parts of this solution. Maybe someone can get some tips with this

pibooth_ledring_start.py
```
"""Plugin to fill the led ring, when entering in the 'wait' state."""
import pibooth
from multiprocessing.connection import Client
from pibooth.utils import LOGGER

__version__ = "1.0.0"

@pibooth.hookimpl

#def state_preview_enter(cfg):
def state_wait_exit():
    LOGGER.info("Fuelle LED ring") 

    address = ('localhost', 3333)
    conn = Client(address)
    conn.send('f')
    conn.close()
```

pibooth_ledring.py
```
"""Plugin to start the led ring, when entering in the 'preview' state."""

import pibooth
#import serial
from pibooth.utils import LOGGER
from multiprocessing.connection import Client

__version__ = "1.0.0"

@pibooth.hookimpl

def state_preview_enter(cfg):
#def state_wait_exit(cfg):
    LOGGER.info("Start LED ring") 
    LOGGER.info(cfg.get('WINDOW', 'preview_delay')) 

    address = ('localhost', 3333)
    conn = Client(address)
    conn.send('s'+cfg.get('WINDOW', 'preview_delay'))
    conn.close()
```

server.py
```
#!/usr/bin/env python
from multiprocessing.connection import Listener
import serial
import re
import logging
import sys

logging.basicConfig(stream=sys.stdout, level=logging.DEBUG)

ser = serial.Serial('/dev/ttyUSB0', 9600)
address = ('localhost', 3333)
listener = Listener(address)
while True:
  conn = listener.accept()
  logging.info('connection accepted from %s', listener.last_accepted)
  while True:
    try:
      msg = conn.recv()
    except EOFError as e:
      logging.info('Connection was closed.')
      break
    if msg == "f":
      logging.info('got F')
      ser.write(b'f')
    match = re.fullmatch("s[0-9]+", msg)
    if match is not None:
    ##if msg == "s1" or msg == "s2" or msg == "s3" or msg == "s3" or msg == "s5":
      logging.info('got an timer: %s', msg)
      ser.write(bytes(msg, encoding='utf-8'))
ser.close()
listener.close()
```

Arduino code
```
#include <Arduino.h>
#include <Adafruit_NeoPixel.h>

#define PIN            6
#define NUMPIXELS      45

Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, PIN, NEO_GRB + NEO_KHZ800);

int shiningDelay;

bool lightAllPixel = false;
bool countdownStart = false;
bool countdownRun = false;
int countdownTimer;

void setup() {
    Serial.begin(9600);
    pixels.begin();
}

void loop() {
  while (Serial.available() > 0) {
    int incomingByte = Serial.read();

    if (incomingByte == 102) {
      lightAllPixel = true;
      countdownStart = false;
      countdownRun = false;
    }
    else if (incomingByte == 115) {
      countdownStart = true;
      lightAllPixel = false;
      countdownRun = false;
    }
    else if (countdownStart == true && incomingByte >= 49 && incomingByte <= 57) {
      countdownRun = true;
      countdownTimer = incomingByte - 48;
      countdownStart = false;
    }
    else {
      countdownStart = false;
      lightAllPixel = false;
      countdownRun = false;
    }
  }

  if (lightAllPixel) {
      // Lite up all LEDs
      for(int i=0;i<NUMPIXELS;i++){
        pixels.setPixelColor(i, pixels.Color(0,20,0));
      }
      pixels.show();
      lightAllPixel = false;
  }
  if (countdownRun) {
    countdownRun = false;
    countdownStart = false;

      shiningDelay = (float(float(countdownTimer) / float(NUMPIXELS)) * 1000) - 1;
      for(int i=NUMPIXELS-1;i>=0;i--){
        for(int dimming = 20-1;dimming >= 0;dimming--){
          pixels.setPixelColor(i, pixels.Color(0, dimming,0));
          pixels.show(); // This sends the updated pixel color to the hardware.
          delay(shiningDelay/20);
       }
      }
  }
}
```

anxuae, 2023-05-06:
Hi @tinni123 

I'm closing this issue because handling USB ring light will no be part of the pibooth core.
An external plugin could be developed but code will depend on the  usb ring light you are chosen.

closed 2023-05-06 by anxuae

karlzre, 2024-11-19:
hi,

@Cruiser79 could you help me to do more or less teh same as you did .
I used in the past the touchselfie app.
I was able to hardcode in the program to send a ser.write ('6') to initiate a neopixel countdown and ser.write ('7') to initia a flash ( all led on wite) 
as 6/7 are related to a do while in the arduino code.

i tested partially your code but not sure how it work.

I created the server.py file and stert it with python3 server.py ( manually for testing) 
I created the pibooth_ledring.py 
`"""Plugin to start the led ring, when entering in the 'preview' state."""

import pibooth
#import serial
from pibooth.utils import LOGGER
from multiprocessing.connection import Client

__version__ = "1.0.0"

@pibooth.hookimpl

def state_preview_enter(cfg):
#def state_wait_exit(cfg):
    LOGGER.info("Start led ring") 
    LOGGER.info(cfg.get('WINDOW', 'preview_delay')) 

    address = ('localhost', 3333)
    conn = Client(address)
    conn.send('6')
    conn.close()`

i just need to sen a 6 for the countdown which will be writed to the neopixel from 3-0 and send a 7 for the flash.
Can you help me how to do that ?

Do you have to add something in the config file ?