pibooth/pibooth #370: Pibooth regulary shutdown (crash)

TheSuperpapi, 2022-07-10:
Hello,

Long time I not reply or say hello there, covid was hard for me...
I'm just taking care for a new time about pibooth and my project.

I start making test and use it for familly party and had issue of crashing, after updating to the last available 3 weeks ago, I still have the same issue.

- Pibooth redulary crash (close) itself after processing screen and after print - 
It is not automatic, sometime it can work 1 hour sometime less but when it become to crash, It crash regulary (every 1 to 5 run).
Between each crash I reboot the Rpi 4.

I don't know how I can send you log file or something else that could help you to explain me what happen.
I don't know if someone had this ussie.

Fill free to ask me what you whant, I'll provide you (but please, just explain me how to find what you could need)

"Maybe it is just an hardware issue..."

Regards to all of you,
Papi

TheSuperpapi, 2022-07-13:
From pibooth.log (but do I need to copy and past the log file just after crash?)

[ INFO    ] pibooth           : Installed plugins: sound-effects-0.0.3
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ DEBUG   ] pibooth           : Try to kill process 'gvfs-gphoto2-volume-monitor'
[ DEBUG   ] pibooth           : Try to kill process 'gvfsd-gphoto2'
[ DEBUG   ] pibooth           : Found gPhoto2 cameras on ports: 'usb:001,005'
[ INFO    ] pibooth           : Configuring hybrid camera (Picamera + gPhoto2) ...
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Lens Name' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'ISO Auto Hi Limit' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'High ISO Noise Reduction' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Live View AF Mode' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Flash Exposure Compensation' expected: 0x0002 got: 0x0001
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Auto White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Tungsten White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Fluorescent White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Daylight White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Flash White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Cloudy White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Shady White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth           : Getting option actions/viewfinder=0
[ DEBUG   ] pibooth           : Setting option imgsettings/iso=800
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Lens Name' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'ISO Auto Hi Limit' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'High ISO Noise Reduction' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Live View AF Mode' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Flash Exposure Compensation' expected: 0x0002 got: 0x0001
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Auto White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Tungsten White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Fluorescent White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Daylight White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Flash White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Cloudy White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Shady White Balance Bias' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : ptp_usb_getresp [usb.c:514]: PTP_OC 0x1016 receiving resp failed: PTP Access Denied (0x200f)
[ DEBUG   ] pibooth.gphoto2   : _set_config [config.c:9101]: 'ptp_generic_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Access Denied (0x200f)
[ DEBUG   ] pibooth.gphoto2   : gp_context_error: The property 'ISO Speed' / 0x500f was not set (0x200f: PTP Access Denied)
[ DEBUG   ] pibooth.gphoto2   : gp_camera_set_config [gphoto2-camera.c:1115]: 'camera->functions->set_config (camera, window, context)' failed: -1
[ ERROR   ] pibooth           : Unsupported option imgsettings/iso=800 ([-1] Unspecified error), configure your DSLR manually
[ DEBUG   ] pibooth           : Setting option settings/capturetarget=Memory card
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Lens Name' expected: 0x0002 got: 0x0004
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'ISO Auto Hi Limit' expected: 0x0001 got: 0x0002
[ DEBUG   ] pibooth.gphoto2   : _get_config [config.c:8708]: Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002
[... cut ...]
[ INFO    ] pibooth           : Connected to printer 'Dai_Nippon_Printing_DP-DS620'
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 41 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 62 8192
[ DEBUG   ] pibooth           : took 334991.428 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 41 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 62 8192

TheSuperpapi, 2022-07-14:
From pibooth.log (but do I need to copy and past the log file just after crash?
[pibooth2.log](https://github.com/pibooth/pibooth/files/9113243/pibooth2.log)

TheSuperpapi, 2022-07-14:
Hi again,

Pibooth ran for 4 hour with a bot without crash.
Impossible to reproduce the crash @ home.

The only difference is that I don't use printer @ home for economic reason (DNP DS620)

Someone think that it can be due to printer job or to big queue or something similar?

djsliders, 2022-07-15:
Hi,
I had the same issue once. No issue during hours, then it was crashing regularly even after reboot.
I have the DNP DS620 too.
Just a curious question, what bot do you use ?
Thanks

TheSuperpapi, 2022-07-15:
Hello,

I'm using -OG Auto Clicker- on windows via vnc from windows workstation.

Yes it seems you have thé same issue as me...
Don't know how to solve it, maybe to limite printing to 2, I'll have to try but will wait a bit to sée if someone Can help finding a Real solve route.

😉

djsliders, 2022-07-15:
This was the log I had after few hours, then repeated itself:
It was working well but then even after a reboot I had this issue.

Actually, the first thing I did was to put the log at a place where it isn't deleted when the raspberry crashed... so I can have the log after the reboot. I don't get why they put it in /tmp/ :/ 
And I don't know why they don't put the timestamp in the log neither... :-P

Thanks for the autoclicker, It is true that without the printing in the stress test, it may be difficult to reproduce the bug.
I will try and keep you up to date.

```
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x27de6f0 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 33, in <module>
    sys.exit(load_entry_point('pibooth', 'console_scripts', 'pibooth')())
  File "/home/pi/pibooth/pibooth/booth.py", line 496, in main
    app = PiApplication(config, plugin_manager)
  File "/home/pi/pibooth/pibooth/booth.py", line 131, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 55, in _multicall
    gen.send(outcome)
  File "/home/pi/pibooth/pibooth/plugins/camera_plugin.py", line 26, in pibooth_setup_camera
    cam = camera.find_camera()
  File "/home/pi/pibooth/pibooth/camera/__init__.py", line 27, in find_camera
    rpi_cam_proxy = get_rpi_camera_proxy()
  File "/home/pi/pibooth/pibooth/camera/rpi.py", line 31, in get_rpi_camera_proxy
    return picamera.PiCamera()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 512, in _init_preview
    self._preview = PiNullSink(
  File "/usr/lib/python3/dist-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2210, in enable
    mmal_check(
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Failed to allocate required memory.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
mmal: mmal_vc_port_enable: failed to enable port vc.null_sink:in:0(OPQV): ENOSPC
mmal: mmal_port_enable: failed to enable connected port (vc.null_sink:in:0(OPQV))0x132b440 (ENOSPC)
mmal: mmal_connection_enable: output port couldn't be enabled
```

tommeuh, 2022-07-16:
Hello,
I also have a DS620 and I also have a random pibooth crashing issue. I managed to reproduce the crash at home. Here is the error, which seems to come from our DS620, no ?
`[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DP-DS620 processing
Fatal Python error: PyEval_RestoreThread: NULL tstate
`[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DP-DS620 processing
Fatal Python error: PyEval_RestoreThread: NULL tstate

Thread 0x74bf2440 (most recent call first):
  File "/usr/lib/python3.7/socket.py", line 589 in readinto
  File "/usr/local/lib/python3.7/dist-packages/cups_notify/listener.py", line 29 in get_chunk_size
  File "/usr/local/lib/python3.7/dist-packages/cups_notify/listener.py", line 56 in do_PUT
  File "/usr/lib/python3.7/http/server.py", line 414 in handle_one_request
  File "/usr/lib/python3.7/http/server.py", line 426 in handle
  File "/usr/lib/python3.7/socketserver.py", line 720 in __init__
  File "/usr/lib/python3.7/socketserver.py", line 360 in finish_request
  File "/usr/lib/python3.7/socketserver.py", line 347 in process_request
  File "/usr/lib/python3.7/socketserver.py", line 316 in _handle_request_noblock
  File "/usr/lib/python3.7/socketserver.py", line 237 in serve_forever
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x78eff440 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 539 in held
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x798ff440 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 539 in held
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x7a2f8440 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 539 in held
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Current thread 0xb6f8db40 (most recent call first):
  File "/usr/local/lib/python3.7/dist-packages/pibooth/printer.py", line 130 in get_all_tasks
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/view_plugin.py", line 153 in state_print_enter
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_callers.py", line 39 in _multicall
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 324 in <lambda>
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_result.py", line 33 in from_call
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 324 in traced_hookexec
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_manager.py", line 80 in _hookexec
  File "/usr/local/lib/python3.7/dist-packages/pluggy/_hooks.py", line 265 in __call__
  File "/usr/local/lib/python3.7/dist-packages/pibooth/states.py", line 91 in set_state
  File "/usr/local/lib/python3.7/dist-packages/pibooth/states.py", line 64 in process
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 402 in main_loop
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 491 in main
  File "/usr/local/bin/pibooth", line 8 in <module>
Abandon
`

rrofa, 2022-07-19:
I have heard of the same issue but was only now able to reproduce it by hammering on the printing button many times (queue was at 25 images left or so). Here is my log file (no debug mode, unfortunately) from before the crash which looks similar to @tommeuh. Fyi: I don't have a DS620 but a CZ-01 which appears to run on similar drivers.

`[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : <Swig Object of type '_CameraWidget *' at 0x925805e0>
[ INFO    ] pibooth           : Take a capture
[ INFO    ] pibooth           : <Swig Object of type '_CameraWidget *' at 0x924fb4e0>
[ INFO    ] pibooth           : <Swig Object of type '_CameraWidget *' at 0x924fb4e0>
[ INFO    ] pibooth           : <Swig Object of type '_CameraWidget *' at 0x924f79c0>
[ INFO    ] pibooth           : <Swig Object of type '_CameraWidget *' at 0x92517060>
[ INFO    ] pibooth           : Saving raw captures
[ INFO    ] pibooth           : Creating the final picture
[ INFO    ] pibooth           : Use TemplatePictureFactory to create background
[ INFO    ] pibooth           : Use TemplatePictureFactory to concatenate images
[ INFO    ] pibooth           : Use TemplatePictureFactory to assemble final image
[ INFO    ] pibooth           : Use TemplatePictureFactory to draw texts
[ INFO    ] pibooth           : Save image '/home/pi/Pictures/pibooth/2022-07-19-21-59-03_pibooth.jpg'
[ INFO    ] pibooth           : Send final picture to printer
[ INFO    ] pibooth           : Send final picture to printer
[ INFO    ] pibooth           : Print Job: CITIZEN_SYSTEMS_CZ-01-823 (2022-07-19-21-59-03_pibooth.jpg) held
[ INFO    ] pibooth           : Printer: CITIZEN_SYSTEMS_CZ-01 processing
[ INFO    ] pibooth           : Send final picture to printer
Fatal Python error: PyEval_RestoreThread: the function must be called with the GIL held, but the GIL is released (the current Python thread state is NULL)
Python runtime state: initialized

Thread 0x8ef28440 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 620 in _blink_device
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Current thread 0x8c907440 (most recent call first):
  File "/home/pi/pibooth/pibooth/printer.py", line 129 in get_all_tasks
  File "/home/pi/pibooth/pibooth/printer.py", line 73 in _on_event
  File "/usr/local/lib/python3.9/dist-packages/cups_notify/listener.py", line 124 in notify
  File "/usr/local/lib/python3.9/dist-packages/cups_notify/listener.py", line 70 in do_PUT
  File "/usr/lib/python3.9/http/server.py", line 415 in handle_one_request
  File "/usr/lib/python3.9/http/server.py", line 427 in handle
  File "/usr/lib/python3.9/socketserver.py", line 720 in __init__
  File "/usr/lib/python3.9/socketserver.py", line 360 in finish_request
  File "/usr/lib/python3.9/socketserver.py", line 347 in process_request
  File "/usr/lib/python3.9/socketserver.py", line 316 in _handle_request_noblock
  File "/usr/lib/python3.9/socketserver.py", line 237 in serve_forever
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x8ffff440 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 539 in held
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x909ff440 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 539 in held
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x9138c440 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 539 in held
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0xb6fb9980 (most recent call first):
  File "/home/pi/pibooth/pibooth/booth.py", line 404 in main_loop
  File "/home/pi/pibooth/pibooth/booth.py", line 491 in main
  File "/usr/local/bin/pibooth", line 33 in <module>
Aborted`

TheSuperpapi, 2022-07-20:
Hello all of you,

Happy that i'm not alone on this issue.
I plan week 33 to try to put 1 automatic printing and limit to a maximum of one more by pressing the option.

Hope this crash will not appear on this way.
Back to you after this test.

tommeuh, 2022-07-20:
Hello @TheSuperpapi & @rrofa 
I have already tried setting pibooth to one print per photo. I still have this crash. I don't have the level to fix the bug, meanwhile I coded a small python script that detects if the pibooth process is open. if it crashed, the raspberry reboot. It's not perfect but in the meantime, it avoids rebooting manually.

```
import psutil
from time import sleep
from os import system
import datetime

currentime = datetime.datetime.now()
hours = currentime.strftime('%H')
minutes = currentime.strftime('%M')
seconds = currentime.strftime('%S')
error_time = currentime.strftime('%H:%M:%S')

reboot = True
process_list = []
error_file = '/home/pi/.config/pibooth/scripts/log_time_errors.txt'

while reboot:
    sleep(20)
    process = psutil.process_iter(attrs=None, ad_value=None) # Liste des process en cours
    for p in process:
        process_list.append(p) #On crée une liste de process
    if "pibooth" in str(process_list): # Si pibooth n'est pas dans la liste
        print("c'est ouvert")
    else:
        print("C'est fermé")
        with open(error_file, 'a') as f: # On log l'heure du crash
            f.write(f"Reboot > {error_time}\n")
        reboot = False
        system("sudo reboot")
sleep(20)    
system("sudo reboot") # Si on quitte la boucle (erreur inconnue) on reboot
```

TheSuperpapi, 2022-07-20:
Hello,
Good news if auto restart is good for you.

On my side when I have the first crash (taking a bit of Time) this crash back after each rpi reboot.

So restart is not a solution on my side and unfortunatly I need to stop all my rent planning... ☹️

(Initialy it was build for m'y wedding but with a rent goal and qualité)

[image]

tommeuh, 2022-07-20:
@TheSuperpapi the commande 
pibooth --regen 
reset your program properly?

TheSuperpapi, 2022-07-20:
Thanks @tommeuh 

Was not so hard for me, I am daily using CAD tools like Catia in my job.
After this is just design Choices, material Choices, machinings and assembly 😁

Was more difficult for me to start learning rpi and python to be able to personalize pibooth (touchscreen interface, backgrounds...)

But it's not the topic of this bug report post 😁😁😁

djsliders, 2022-07-23:
Wow I really like your photobooth ! Can we discuss in PM ?
I can help you with python and some technical stuff too ;-)

TheSuperpapi, 2022-07-23:
Hello, 

Yes we can talk pm @djsliders and in french if you come from the same country as me.

For helping, you can maybe help us on this issue?
If not, you can still pm me 😉

djsliders, 2022-07-23:
I got exactly the same issue this morning :-/ 

Btw, I added the timestamp in my logs, and move them from /tmp, if you want help for that.

Concerning the issue, maybe we should try to add a try / catch around : 
 File "/home/pi/pibooth/pibooth/printer.py", line 129 in get_all_tasks

TheSuperpapi, 2022-07-23:
You can catch me on discord: SPG_Superpapi#4036 @djsliders

clederman, 2022-07-27:
Hello,

I encounter the same problem with pibooth 2.0.4 or 2.0.5 on a RPI3 with a dnp DS40 printer.

TheSuperpapi, 2022-07-27:
Seems there is an issue with ds printer.
My printer is into a box and crashing after 1h and after first crash it's crashing most of the Time even if I reboot the rpi.
I don't know if it could due to printer overhead and if it is the case, how it Can crash pibooth...

anxuae, 2022-08-04:
Hi @tommeuh do you have some lines before the crash that you have posted?

> [ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DP-DS620 processing
> Fatal Python error: PyEval_RestoreThread: NULL tstate
> Thread 0x74bf2440 (most recent call first):

Hi @rrofa, the logs seems to indicate that you have triggered 3 time quickly the "print button", is it voluntary ? If not, are you using physical button? In that case I invite you to change the `debounce_delay` in the config and try again.

> [ INFO ] pibooth : Use TemplatePictureFactory to draw texts
> [ INFO ] pibooth : Save image '/home/pi/Pictures/pibooth/2022-07-19-21-59-03_pibooth.jpg'
> [ INFO ] pibooth : Send final picture to printer [...]

Globally, I think that this crash is coming from the [pycups](https://github.com/OpenPrinting/pycups) library which is not enough robust to to parallel actions.

rrofa, 2022-08-04:
Hi @anxuae,

no, I just tried to force the crash by hitting the printing button many times just by clicking on it. I came accross the same issue without doing it that often, as well. Maybe, setting an idle time after the print button is pushed is a solution? I figured that people usually hit the button twice before they realize that the printer is already printing and then get a second print that wasn't intended.
I hacked a quick fix into my code quite rudimentarily but can't yet say if it solves the problem. If someone is interested, I can post the changes here later. The basic idea is to define the idle time and check each time, the print button is pushed, if enough time has passed since.

tommeuh, 2022-08-04:
Hello @anxuae thanks for your help. Yes, before this lines y've classics [INFO] lines, if we need I can try to force the crash again to get all the lines. you think the bug is not the same for me as for @rrofa ?

TheSuperpapi, 2022-08-04:
Hello all,
Seems this is the same issue as me.
I'll try to reproduce the bug, but it only happen when printing (touchscreen) and true that poeple are hitting 2 or 3 Time before to see this is printing.

I'll have also to check if it's not dur to a température issue but don't understand how printer température could crash thé software.

I'll follow all your proposal and make some test when I'll back from vacations.

clederman, 2022-08-06:
Hello,

I reproduced the problem during a party.
Maximum number of prints set to 4.
A first press on print, printing is in progress and a second press on print, and the pibooth crashes.
A relaunch of the Pibooth process and here we go again.

I was unable to reproduce the crash afterwards.

Here is the trace of the logs:

> [ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 1261 8192
[ INFO    ] pibooth           : Print Job: Dai_Nippon_Printing_DS40-1211 (2022-07-31-02-38-14_pibooth.jpg) held
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing
[ DEBUG   ] pibooth           : BUTTONDOWN: generate PRINTER event
[ INFO    ] pibooth           : Send final picture to printer
[ INFO    ] pibooth           : Printer: Dai_Nippon_Printing_DS40 processing

anxuae, 2022-08-07:
Hi @clederman do you have the end of the log?

anxuae, 2022-08-08:
Hi guys,

Can you try if the bug is still present after upgrading pycups-notify ?

`pip3 install -U pycups-notify`

dominiquegrota, 2022-08-09:
Hi, 
I confirm I get the same error, It only happens while printing and taking new pictures... Very difficult to reproduce, because it can happend every 100 prints, or twice in 10 prints.

I did not yet try to upgrated pycups-notify. I will give you feedback when tested.

Thank's

clederman, 2022-08-11:
> Hi @clederman do you have the end of the log?

Hi @anxuae, it's the end of the log.
I had the same crash with an identical end of log
[pibooth.log](https://github.com/pibooth/pibooth/files/9307943/pibooth.log)

I tested 

> pip3 install -U pycups-notify.

but unable to update, here is the log:

> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pycups-notify
  Downloading https://files.pythonhosted.org/packages/fb/24/daba6c51c4fab7e51abafaa6f7c6e5991219fdffc80d25e7eb0a47530aee/pycups_notify-0.0.6-py2.py3-none-any.whl
Requirement already satisfied, skipping upgrade: pycups>=1.9.73 in /usr/lib/python3/dist-packages (from pycups-notify) (1.9.73)
Installing collected packages: pycups-notify
Successfully installed pycups-notify-0.0.6

tlacassi, 2022-08-21:
Hello ! I'm using a pi 4 8gb, a canon 700d and a DNP qw410. I also happens to have random crashes. Sometimes after 10 photos, and sometimes after 100. Cannot reproduce the bug myself

TJay0, 2022-08-21:
Hi All,

I'm using a Nikon D3200 and a Citizen CZ-01 and I have the same issues as well. As already discussed it looks like a problem with the printer driver or with Cups. At my last photobooth event the pibooth crashes just in time I hit the printbutton on the touchscreen. I am not a software expert but it seems very difficult to fix a driver problem. Maybe a automatic restart of the software due to a crash would be the easier quick solution but I dont know how to implement it on the rpi :) 

Greetings

TheSuperpapi, 2022-08-22:
> Hi All,
>
> I'm using a Nikon D3200 and a Citizen CZ-01 and I have the same issues as well. As already discussed it looks like a problem with the printer driver or with Cups. At my last photobooth [...]

Hello Tjay,
You can look at aboce, there is a post from [tommeuh] with the code.
Maybe you can ask him to explain how to implement it. 

See you.

anxuae, 2022-08-25:
Thread issue found in pycups. https://github.com/OpenPrinting/pycups/issues/43

anxuae, 2022-08-27:
Hi guys,

New version 2.0.6 available. I've tried to implement a fix. Can you try to reproduce?

Exitras, 2022-08-31:
Hi all, 
I was experiencing the same issue of random shutdowns when printing was involved and just updated my booth to 2.0.6. So far, the booth mastered my first stress test where I randomly pressed all buttons, printed 15 images in a row, and shot countless pictures with 1sec preview delay.

However, I'm still experiencing another issue that causes random shutdowns. My camera (Canon 2000D) sometimes turns off by itself and requires me to restart the entire booth. 
This can be seen in the logs (please also find the full logs file attached):

```
gphoto2.GPhoto2Error: [-10] Timeout reading from or writing to the port
[ DEBUG   ] pibooth           : took 60.010 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.103-v7l+', version='#1529 SMP Tue Mar 8 12:24:00 GMT 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
```

Does any of you experience the same issue?

My setup is as following:
Raspberry Pi: RPi 4 with 4Gb ram.
Cooling fan on Raspberry Pi: yes
Original RPi USB-C PSU: yes
Current Software Version: 2.0.6 (same issues with 1.x.x and 2.x.x)
PiBooth LED feature used: No
Pictures additionally stored on USB: Yes
Touchscreen: No
Camera: Canon 2000D
Camera Power: Stepdown-converter LM2596 on 7,2V with 3A and dummy battery.
Second camera: No
Autofocus on camera: Off
Camera mode: Manual
External flash: No
Ferrite bead on camera connection cable: Yes
Printer: DNP DS620

(I described this issue in 2020 already: https://github.com/pibooth/pibooth/issues/182)

Cheers, 
Exitras

[pibooth.log](https://github.com/pibooth/pibooth/files/9461722/pibooth.log)

froehnerson, 2022-08-31:
Hey hey,

I was experiencing the same issue of random shutdowns when printing was involved like @Exitras. The stress test with 2.0.6. is still pending, however i also have the same problem of random shutdowns with the Canon 2000D which sometimes turns off by itself and requires me to restart the entire booth.

Also otherwise I have pretty much the same setup as @Exitras.

Since I had a similar problem as in issue "can not get camera preview image after first photo #388" with the Nikon d5300, I have the following question, which camera do you use and do you have the problem of random shutdowns of the camera too?

Cheers

anxuae, 2022-09-03:
Hi @froehnerson ,

You have to setup your  DSLR with `Auto Power Off` = false. You should be able to set it in your DSLR menu.
Some example of DSLR used here: https://github.com/pibooth/pibooth/tree/master/tests/dslr_diag

TheSuperpapi, 2022-09-10:
> Hi guys,
> 
> New version 2.0.6 available. I've tried to implement a fix. Can you try to reproduce?

Hello Anxuae,

I'll try to make a Real test before the end of this mounth in a Friend party.
I'll keep you UP yo date as soon as I can.

😉

TheSuperpapi, 2022-09-10:
Testing 2.0.6

Full RPI freeze after this screen... strange

[image]

TheSuperpapi, 2022-09-10:
Last post with crash information removed (installed 64 bits version....... ok on 32 for sure)

tlacassi, 2022-09-10:
I installed 2.0.6, tested it at a party for 5h straight. Not a single crash anymore ! Even with the drunk users !!!

TheSuperpapi, 2022-09-11:
Songs good 😁

TheSuperpapi, 2022-09-11:
Just had issue with cups and DNP ds620.
Brand not in the list - installation impossible for me, could help me?

[image]

TheSuperpapi, 2022-09-11:
Issue solved with: sudo apt-get install printer-driver-gutenprint

Exitras, 2022-09-13:
Hi all, I tested it last weekend at a small birthday party where 93 pictures have been taken (most of them also printed). We had not a single issue. I'll keep you updated but so far, the fix seems to work well :)

clederman, 2022-09-14:
Hello,
I applied the patch and tested during 2 evenings, around 130 photos printed per evening.
No crash observed, with the option 1 photo duplication and the other with 4 photo duplications.
I think the fix is working.
A big thank's.
I watch the next uses.

grafbrueckner, 2022-09-18:
Hi all, 
thanks for your efforts. I had multiple crashes on my own wedding. I'm looking forward to test the fix next week at another party ;)

TheSuperpapi, 2022-09-25:
Hello, coucou,

I have tried the patch Yesterday on a friend event for 658 print.

I had no crash 😁 (just some freeze at the end of the preview counter but no link)

So, I propose a definitive issue solved switch.

Thanks again a lot 😉👍

anxuae, 2022-09-28:
Thanks all for your feedbacks, this is a really good news!

grafbrueckner, 2022-09-28:
@TheSuperpapi the same here at my party. It sometimes freezed after the preview counter but no restart was needed :)