pibooth/pibooth #371: Show Printer state

Ulmerju, 2022-07-15:
Everything works fine until my printer runs out of paper or ink

The status of the printer goes to "En pause - "Printer error: No Paper (03)"

Even if you put paper in, turn the printer off and on again, the status in CUPS doesn't change and it stops printing.
I have to manually click "Start printer" in the CUPS interface, then the printer works fine.

On the one hand, it would be nice:
- if the current error messages, e.g. no paper or ink can be ejected directly. So the operator sees this without closing the program and looking in cups.

On the other hand, it would be ideal:
- if the print job is restarted automatically as soon as the paper or ink has been refilled.
The main problem I see here is that the printers only return the correct message after they have been restarted from the pause.

A number of people have encountered this problem:
https://github.com/OpenPrinting/cups/issues/278
https://github.com/pibooth/pibooth/issues/296

nicohofdtz, 2022-07-18:
I thought I had the same issue before. I just wanted to investigate this issue and find a possible solution but when I tested it my CP1300 continued to print after I refilled paper without any issue.
As far as I can tell the fix mentioned in #296 has not been implemented. According to the git history no work has been done to printer.py for 2 years.

Probably the newer printer just behaves different but if I can be of any further help I will be happy to do so.

Still weird that I am pretty sure, I had the issue myself before. I have a wedding coming up next weekend. I will see (and report) if I have any issues over the course of the day.

dominiquegrota, 2022-08-09:
> 

Hi, I had the same problem you encoutered, so I added a command in the printer.py to activate cups. So fi cups is stopped, it is automatically activated before to get to the printing section.

`        os.system("cupsenable Canon_SELPHY_CP1300")`

I also added a paper sensor on my printer on gpio 23. The rapsberry tests if the sensor is off or a printer error exists, then the max-page is set to 0. This avoids to promt the "print" message if somthing gets wrong.

```
printer_status = os.popen('lpstat -p').read()
               if printer_status.find("Printer error:")<0 and GPIO.input(23) == 0:
               self.max_pages = -1
        else:
               self.max_pages = 0
```
But all this trouble is only because of the canon printer :-(

Dominique

nicohofdtz, 2022-08-09:
Very weird that it works for me but not for you although we have the same printer.
I had no issues at the above mentioned wedding. I changed paper multiple times and also ink once and the printer always resumed printing automatically after a few seconds.

dominiquegrota, 2022-08-09:
Thank's for your answer @nicohofdtz, 
do you use your cp1300 with usb or wifi connected ? I use usb. May be the printer gives paper status when using wifi ?

nicohofdtz, 2022-08-09:
I indeed used the printer over wifi. And you are very likely correct as the drivers are different.

anxuae, 2022-08-09:
Hi @dominiquegrota 

Can you tel me if replacing `os.system("cupsenable Canon_SELPHY_CP1300")` by `self._conn.enablePrinter(self.name)` to reenable printer is working?

dominiquegrota, 2022-08-28:
Hi @anxuae, 
Sorry for myl late response :-/

I just tried to replace os.system("cupsenable Canon_SELPHY_CP1300") by self._conn.enablePrinter(self.name)  in printer.py

I get an error : TypeError : unicode or bytes object required

Dominique

anxuae, 2022-08-29:
May I ask you to retry with this line?

```python
self._conn.enablePrinter("Canon_SELPHY_CP1300")
```

dominiquegrota, 2022-08-29:
of couse you can !
`self._conn.enablePrinter("Canon_SELPHY_CP1300")` worked for me exactly as `os.system("cupsenable Canon_SELPHY_CP1300")`

anxuae, 2023-05-06:
Enabling printer will be tried in v3.0.0, coming soon

closed 2023-05-06 by anxuae