pibooth/pibooth #377: CPU Usage

jo-ei, 2022-07-18:
**Describe the bug**
CPU core with 100% without showing an live image oder anything else as the start screen

the cpu frequency of the raspi then also increases from 700MHz to 1500 MHZ which makes it very warm

**Steps to reproduce the bug**
1. start pibooth
2. open htop

**Environment**
 - Debian 10.12
 - Camera [Canon EOS 1200D]
 - Pibooth git version [2.0.5]

[image]

[image]

jo-ei, 2022-07-29:
I think I was able to fix the problem by setting the GPU memory to the max value. Let's hope

anxuae, 2022-08-04:
Hi @jo-ei 

Can you tell me what is your resolution put in the configuration?
Are you also using a Picamera (hybrid mode)?
Which plugin are you using?
Did you set specific window background?
Did you set picture background/overlay?

The size of the capture/background/overlay to process may imply a big cpu load.

jo-ei, 2022-08-05:
Hi @anxuae,
thansk for your help!

>Can you tell me what is your resolution put in the configuration?
1280x1024

>Are you also using a Picamera (hybrid mode)?
No, only a Canon Eos D1200

>Which plugin are you using?
for test to reproduce: none

>Did you set specific window background?
not set knowingly

>Did you set picture background/overlay?
not set knowingly

```
pi@pi4:~ $ pibooth -v
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins:
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ DEBUG   ] pibooth           : Found gPhoto2 cameras on ports: 'usb:001,004'
[ INFO    ] pibooth           : Configuring gPhoto2 camera ...
[ DEBUG   ] pibooth           : Getting option actions/viewfinder=2
[ DEBUG   ] pibooth           : Setting option imgsettings/iso=100
[ WARNING ] pibooth           : Invalid value '100' for option iso (possible choices: ['Auto']), trying to set it anyway
[ DEBUG   ] pibooth.gphoto2   : ptp_usb_getresp [usb.c:514]: PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
[ DEBUG   ] pibooth.gphoto2   : _set_config [config.c:9128]: 'ptp_canon_eos_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Device Busy (0x2019)
[ DEBUG   ] pibooth.gphoto2   : gp_context_error: The property 'ISO Speed' / 0xd103 was not set (0x2019: PTP Device Busy).
[ DEBUG   ] pibooth.gphoto2   : gp_camera_set_config [gphoto2-camera.c:1115]: 'camera->functions->set_config (camera, window, context)' failed: -110
[ ERROR   ] pibooth           : Unsupported option imgsettings/iso=100 ([-110] I/O in progress), configure your DSLR manually
[ DEBUG   ] pibooth           : Setting option settings/capturetarget=Memory card
[ WARNING ] pibooth           : No printer found (pycups or pycups-notify not installed)
[ DEBUG   ] pibooth           : +< Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: |  b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: |  b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: |  STREAM b'IDAT' 140 5756
```

jo-ei, 2022-08-06:
add files
[diagnostic.log](https://github.com/pibooth/pibooth/files/9275068/diagnostic.log)
[pibooth.cfg.txt](https://github.com/pibooth/pibooth/files/9275070/pibooth.cfg.txt)

anxuae, 2023-05-06:
Hi,

I'm closing this old issue because no more occurence.

jo-ei, 2023-05-08:
the problem still exists