pibooth/pibooth #378: Getting a square output image

etieng, 2022-07-24:
First, thanks for all the work done, it has been a really good time using pibooth at the past new years eve !

I'm playing around and wish to go for an output **square** picture.
I've been playing around with virtual printer setup (_cups_pdf_) with custom paper (100x100mm) but didn't changed the output picture.
I've been trying with a mask, but it takes ages to achieve the result and is finally not square (overlay PNG was 360x240 in size)

Do you have any advice for how to change the final picture size and let pibooth crop the captured picture ?

anxuae, 2022-08-03:
Unfortunately the paper size is always set 4 * 6 in

BramWerbrouck, 2022-08-03:
> Hello @etieng
> 
> Unfortunately the paper size is always set 4 * 6 in

Is it possible to change the paper size?

anxuae, 2022-08-03:
As a quick fix, you can change the code here : https://github.com/pibooth/pibooth/blob/8ad42a83e8e89164b14da3f5bf4c591daad686d8/pibooth/pictures/__init__.py#L163

anxuae, 2022-08-04:
Hi @etieng  

Try using the [pibooth-picture-template](https://pypi.org/project/pibooth-picture-template/), you can specify the size of the page in the template.

Please, re-open if you have any issue

closed 2022-08-04 by anxuae