pibooth/pibooth #380: Autostart failed if I set another picture path

andhey, 2022-07-31:
Hello,
for your information. If I adjust the path in the Config so that the pictures are saved on a USB stick, the autostart does not work. When I reset the data path, everything works normally again.

**Environment**
 - OS Raspbian Bullseye
 - Camera Canon EOS 100D
 - Pibooth version 2.05

anxuae, 2022-08-04:
Hi @andhey 

I'm guessing that USB is not yet ready when `pibooth` starts. Can you add a delay before starting (maybe 10s) by editing the startup file as described here https://www.linuxuprising.com/2020/11/how-to-launch-startup-applications-with.html ?

If it's working, I will add an option in the configuration to do it.

Thanks for your feedback

anxuae, 2022-08-04:
Auto start file is located here: `~/.config/autostart/pibooth.desktop`

andhey, 2022-08-07:
Hi @anxuae,

Unfortunately, this does not have the desired effect. Pibooth does not start automatically even when the "delay" is inserted.

andhey

anxuae, 2022-08-09:
Hi @andhey 

Just after your raspberrypi start-up, (with path set to your USB stick), cay you have a look in the `/tmp` directory and look for a `pibooth.log` file? If it's exists, please paste it here.

anxuae, 2022-08-09:
Also, maybe the `X-GNOME-Autostart-Delay` is not recognized on raspberrypi.

In same autostart file, you can try to replace the line `Exec=pibooth` by:

`Exec=bash -c "sleep 15 && pibooth"`

andhey, 2022-08-09:
pibooth.log:

[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B

andhey, 2022-08-09:
`Exec=bash -c "sleep 20 && pibooth"`
works. Perfect. A lot of thanks

anxuae, 2022-08-09:
Merged in master (see #384), new configuration option `autostart_delay` will be available in next pibooth release

closed 2022-08-09 by anxuae