pibooth/pibooth #381: Check camera status in main loop and try reconnect if camera is not found

nicohofdtz, 2022-07-31:
On last weekends weeding I found the majority of the issues with the pibooth could be resolved if the main loop would check if the camera is still there and try a reconnect if it wasn't.
My camera restarted a couple times (I guess due to children taking image after image) and a simple reconnect attempt should have succeeded.

I already tried to implement a clumsy version of my idea: 
I added the _get_summary_ method to the _GpCamera_ class  as it's call fails if the camera is disconnected in contrast to many of the other gphoto methods that will still return if the object is initialized. Next I added a counter to the main loop and used this to check the camera presence once a second. I did this to prevent an overload of the camera connection with this request. I also think a second is plenty.
I used a try-except structure where the reconnect part is in the except body.

Here's the code together with the two original lines of the main loop:
```
pygame.display.update()
check_camera_counter +=1
if check_camera_counter == fps:
    try: 
        self.camera.get_summary()
    except:
        camera_recon = False
        while not camera_recon:
            time.sleep(2)
            try: 
                self.camera.get_summary()
                camera_recon = True
            except:
                LOGGER.debug('Reconnecting camera.')
                try:
                    self.camera.quit()
                    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
                except BaseException as e:
                    LOGGER.debug("Couldn't reconnect")
                    LOGGER.debug(e)
check_camera_counter %= fps
clock.tick(fps)  # Ensure the program will never run at more than <fps> frames per second
```

Working:
When I turn off the camera to simulate a camera crash, pibooth doesn't crash but enters the reconnect mode.

Not Working:
When I turn the camera back on, pibooth doesn't reconnect.
As you can see, I tried to log the error but the output of the third to last line only outputs `_gp_logcb_`.
Shouldn't my code work as expected? Do you have any tips?

anxuae, 2022-08-03:
Hello @nicohofdtz 

Thanks for your question. Ufortinatteny is depending on the camera used, and the implementation that you suggest may lead to a deadlock, and so may be dangerous.

Here is an example of plugin to implement the "retry" process if your camera is not enough robust:

https://pibooth.readthedocs.io/en/latest/sources/plugins/examples.html#setup-a-custom-camera

closed 2022-08-03 by anxuae

nicohofdtz, 2022-10-04:
Sorry for the late reply and thanks for your's.

I'm on the sony branch so if I understand the code you linked correctly that is already implemented.
See: https://github.com/pibooth/pibooth/blob/sony/pibooth/camera/gphoto.py
The problem is that a retry does not reinitialize the connection to the camera. If, as in my case, the camera restarts for some reason, this code will not establish a new connection.
I would really appreciate if pibooth would have routine of trying to reinitiate the connection **before** the oops-message is displayed.

But there is another issue with this retry process. It takes very long till a new attempt to capture is made.
I stopped the time and is more than 30s in my case. I think this is just the time that gphoto2 takes till it times out.
I tried to create a custom timeout using signal.alarm but although the exception was raised the except block was not triggered until gphoto2 timed out also.