pibooth/pibooth #382: Lancement PiBooth impossible

Fabwiss, 2022-08-03:
Bonjour,

Voici le message d'erreur que j'ai à chaque lancement de **PiBooth**
`pi@rpi-pibooth:~ $ pibooth
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 131, in __init__
    self.camera = self._pm.hook.pibooth_setup_camera(cfg=self._config)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 55, in _multicall
    gen.send(outcome)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 26, in pibooth_setup_camera
    cam = camera.find_camera()
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/__init__.py", line 52, in find_camera
    raise EnvironmentError("Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected")
OSError: Neither Raspberry Pi nor GPhoto2 nor OpenCV camera detected
`
Auriez-vous une idée svp ?

Par avance merci pour votre aide.

Fabwiss.

anxuae, 2022-08-04:
Hi @Fabwiss 

Your camera is not detected. Which kind of camera are you using? For Picamera, check that it is enabled in `raspi-config`: 

See: https://raspberrytips.com/install-camera-raspberry-pi/

[image]

Fabwiss, 2022-08-04:
Hello @anxuae,

Thank you for you answer.
I'm french but I try with my (bad) english. Sorry for that ;)

I use a RPI 3 and I have **official HQ Camera** and **official Objectif Grand Angle 6mn**.
In the configuration (raspi-config), I don't have **Camera** but **Legacy Camera**
[image]

I **Enable** and **Yes** on this option.

How can I have **Camera** ?

Fabwiss.

anxuae, 2022-08-04:
Can you run those commands and tell me the output?

`vcgencmd get_camera`

`libcamera-hello`

Fabwiss, 2022-08-04:
Hi,

I managed to connect my camera on my RPI. Good.

Now I have this error. I share my log (pibooth.log) with you : 

[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ DEBUG   ] pibooth           : Try to kill process 'gvfs-gphoto2-volume-monitor'
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ DEBUG   ] pibooth           : Preview resolution is (640.0, 480.0)
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : took 6.219 seconds
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : took 2.728 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[... 26 lines cut ...]
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 69, in state_preview_enter
    app.camera.preview(win)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 155, in preview
    self._window.show_image(self._get_preview_image())
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 101, in _get_preview_image
    raise IOError("Can not get camera preview image")
OSError: Can not get camera preview image
[ DEBUG   ] pibooth           : took 0.030 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='rpi-pibooth', release='5.15.32-v7+', version='#1538 SMP Thu Mar 31 19:38:48 BST 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.040 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'

Do you have an idea to resolve this ?

Fabwiss, 2022-08-04:
**First command :**
> pi@rpi-pibooth:~ $ vcgencmd get_camera
> supported=0 detected=0, libcamera interfaces=0

----

**And the other one :**
> pi@rpi-pibooth:~ $ libcamera-hello
> libEGL warning: DRI2: failed to authenticate
> Made X/EGL preview window
> [0:13:02.888350040] [1399]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
> [0:13:02.992962789] [1414]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
> [0:13:02.994334608] [1399]  INFO Camera camera.cpp:1029 configuring streams: (0) 2028x1520-YUV420
> [0:13:02.995241845] [1414]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  failed to import fd 20
> Abandon

anxuae, 2022-08-04:
It trying to use OpenCV to manage RPI camera

> [ DEBUG ] pibooth : Try to kill process 'gvfs-gphoto2-volume-monitor'
> [ INFO ] pibooth : Configuring OpenCV camera ...
> [ DEBUG ] pibooth : Preview resolution is (640.0, 480.0)
> [ DEBUG ] pibooth : Activate state 'wait'

I think that you have installer PIOS Bulleyes which is not fully supported
Please re-flash your sdcard wil PIOS Buster

Fabwiss, 2022-08-04:
Thank you @anxuae.
I reflash my sd-card with PI OS BUSTER. 
Now I can take photos but....
How can I :

1. show me (or other) on the screen before the photo
2. see the sreen for setup the photos because all my photos are blurred :(

anxuae, 2022-08-05:
Hi

1. If you have black screen, check `iso` value in the config (put an higher value, e.g 500). Other solution is that your camera is not well configured . try to run the `raspistill -t 0` command
2. seems your focus is not correct , it should be configured manually on the lense

anxuae, 2022-08-27:
Some news on this topic?

anxuae, 2022-09-11:
Please re-open if you still have an issue

closed 2022-09-11 by anxuae