pibooth/pibooth #386: Sepia or Vintage Lock of the Photos

andhey, 2022-08-11:
Is there a way to make a filter over the images after taking them so that they look black and white, sepia or vintage?

werdeil, 2022-08-11:
If you are using the picamera you can have a look at the effects: https://pibooth.readthedocs.io/en/latest/sources/start.html?#captures-effects
You may find some cool filter to apply.

andhey, 2022-08-11:
sSrry, im am using a Canon Eos 100d with gphoto2. Does OpenCV have a possible solution?

anxuae, 2022-08-27:
Hi @andhey 

For gphoto2 camera, the possible image effects are defined here: https://pillow.readthedocs.io/en/latest/reference/ImageFilter.html

anxuae, 2022-08-28:
Could be a good idea to implement same effect than picamera using PIL library

andhey, 2022-08-28:
That would be a great idea.