pibooth/pibooth #389: Unsupported option action/viewfinder

GoempieK, 2022-09-02:
When pibooth wants to take a picture I get this error and the application crashes

[ INFO    ] pibooth           : Take a capture
[ ERROR   ] pibooth           : Unsupported option actions/viewfinder=0 ([-7] I/O problem), configure your DSLR manually

 - Camera Canon SX720 HS
 - 
[diagnostic.log](https://github.com/pibooth/pibooth/files/9476458/diagnostic.log)

anxuae, 2022-09-03:
Hi 

Issue seems coming from the communication with the DSLR. Be sure that you have install PiOS Buster and your DSLR autofocus is off.

```
Take a capture
Gphoto2: b'gp_libusb1_read [libusb1.c:614]': b"'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Operation timed out (-7)"
```

GoempieK, 2022-09-09:
Thanks have since used a different camera.
Did not experience any problem with the other camera.

Thanks

closed 2022-09-09 by GoempieK