pibooth/pibooth #390: Add command to reprint previous generated photo.

Kandax86, 2022-09-06:
Hello

I did not find a way to reprint a previous photo stored with a command line. It could be helpful to send people photo after event. I tried with some preview picture software but printing was not possible or can't send data correctly on canon selphy 1300 werhase it works with pibooth.

So I'm looking for a command where I tell a picture path to start printing with same parameters than pibooth normal execution.

I don't know if I'm clear enough 

Thanks for your help.

werdeil, 2022-09-07:
As `pibooth` is only sending the generated picture to CUPS (and CUPS handle the parameters) I guess what you want is already present in CUPS command line tools:
```lp <filename>```
Can you confirm it fill you need?

Kandax86, 2022-09-07:
hello,
thanks for your answer i'll try it, i thought we need to set print paper size resolution & co....

werdeil, 2022-09-07:
The printer configuration is set on CUPS side, we gave an example [here](https://pibooth.readthedocs.io/en/latest/sources/start.html?#printer) but if it worked when using live you shall not have to reconfigure it.

Kandax86, 2022-09-07:
yep worked perfectly you are the "man" :) thank you !!!!

 `Je crois que tu es français alors merci :) beau travail cet appli tu as participé à la réussite de mon mariage ce week end !!! :)`

Kandax86, 2022-09-07:
[image]
[image]

closed 2022-09-09 by werdeil