pibooth/pibooth #391: RAW folder without subfolders

Exitras, 2022-09-08:
**Is your feature request related to a problem? Please describe.**
It would be nice to have all images in the RAW folder without the subfolders for each take as downloading the images would be more convenient.

anxuae, 2022-09-08:
Hi @Exitras 

Thanks for your feedback. However the tool `pibooth-regen` use this raw folder in order to be able to regenerate the final pictures afterwards. And thus, we need to keep the image of a sequence in a separate directory.

closed 2022-09-08 by anxuae

Exitras, 2022-09-13:
Hi @anxuae, thank you for your comment. I totally understand that you need this folder structure for processing the images. However, wouldn't it be possible to move the images to another folder (for example called "originals") after it is processed? It would save me lots of time :)

DJ-Dingo, 2022-09-14:
Maybe a small plugin or script funktion would be an idea "pibooth-copyraw. Where you can select the main folder you want to have copied, and then "copy to a new folder (originals)" which thereby makes a copy from all the separate folders in raw to "originals" within the main folder you have chosen.

werdeil, 2022-09-14:
@Exitras the thing you want can be done with a small command line, like:
```find . -name \*.jpg -exec mv {} newDir \;```
Whereas it is difficult to do the contrary whenever you want to use `pibooth-regen`

A plugin can be done as suggested by @DJ-Dingo but I don't think it is worth doing it inside pibooth (again one single command line to get the wanted folder at the end).