pibooth/pibooth #392: Printer supply level in PiBooth Menu

Exitras, 2022-09-08:
**Describe the solution you'd like**
I'm using the DNP DS620 printer and would love to see the remaining supply level (paper) for example directly in the PiBooth menu. I can currently only guess based on the size of the supply paper roll. I don't know if it is possible to read the printer's supply level with a Pi but it is possible with MacOS or Windows which means that the printer supports this feature.

anxuae, 2022-09-08:
Hi @Exitras 

Thanks for your feedback. I will look if the levels can be retrieved from CUPS server.

TheSuperpapi, 2022-09-16:
Hello Exitras.

If it could help you :

A printing counter has been added in pibooth few Time ago.

It was for this kind of need, and is compatible with all printer.

We just have to think about resetting the counter after the "cartbridge or papier roller" change.

Regards

anxuae, 2023-03-02:
Indicator could be added at screen in case of `printed` counter reached a specific value. But finding the correct value will remain empiric. Thus it is not a perfect solution.

closed 2023-03-03 by anxuae