pibooth/pibooth #393: Touchscreen printing ehancement proposal.

TheSuperpapi, 2022-09-12:
**Is your feature request related to a problem? Please describe.**
When using touchscreen, people don't really understand if the printing is ingoing and how really to print multiple time

**Describe the solution you'd like**
One solution could be to add 2 step
Step one: One screen with printing number choice
Step two: Waiting screen with information: Printing ongoing

The step two "alone" could be a first good to, it'll alow people that printing has been taken into account.

What do you think about?

anxuae, 2022-09-21:
Hi @TheSuperpapi 

Thank you for report. We could make the things more obvious each time the print button is triggered. But have th choice to select the printing number, will reduce the usability of the Photo Booth IMHO.

TheSuperpapi, 2022-09-22:
Hello,

Thanks for the reply.

The waiting screen do not need to wait the end of the print.

But customer need a answer that printing ongoing and this screen could stay 5 to 10s for exemple.

For the number of print I saw that people are pushing 5 or ten Time, because they think that nothing happen and sometime they push only once saying, fuck there is only 1 print, we have to redo.

Don't know what could be the n'est solution to improve that.

dominiquegrota, 2022-11-03:
Hi, 
I wrote a plugin to add a page with "You send your picture to the printer, Your photo will be there in one minute". This page is shown for 5 seconds and it goes back to the wait state. 
I am not a developper and It is probably not the best way to do it, but if you want I can share the code.
Dominique

werdeil, 2022-11-03:
Please share it, it will always help

dominiquegrota, 2022-11-14:
I used this plugin to show the printing notice and it worked for over 230 pictures and 130 printed pictures. I had no bug 😄 

If someone can try it, it could be added in the pibooth-plugin list.

Thank's

TheSuperpapi, 2022-11-15:
Hello,

I think it could interest some of us to improve the user interface.
It had a more visible software respond to user action when pressing to print.

If you could share it as a plugin it could be perfect 😁

Good job 😉

flammeur, 2023-01-10:
Hello,
Is there a pluging available to make waiting screen ?

regards

anxuae, 2023-03-02:
User interface when print is triggered will be improved in v3.0.0

anxuae, 2023-03-17:
Implemented in v3.0.0 coming soon. Hardware button now trigger visible action on user interface

closed 2023-03-17 by anxuae