pibooth/pibooth #395: Kubuntu 20.04 - Fails to capture frame from laptop integrated webcam

samuelgbrown, 2022-09-17:
**Describe the bug**
When attempting to capture a frame using the integrated webcam on a Kubuntu laptop, the app will soft-crash (to fail-safe) during capture state _despite successfully showing a camera preview during the preview state_.

**Steps to reproduce the bug**
1. Start pibooth on described system
2. Click "Take Photo" -> "1 Photo"
3. Observe functioning camera feed during "preview" phase.
4. At end of count-down, observe "Oops, something went wrong"
5. Soft crash to main menu

**Secondary bug behavior:**
When a DSLR camera (Canon Rebel XS) is plugged in, the preview will show the output of the integrated webcam, but the capture state successfully captures a frame from the DSLR (no crash).

The `pibooth-diag` function appeared to fail to properly communicate with the camera, but the diagnostics log is attached below.

**Logs**
[pibooth.log](https://github.com/pibooth/pibooth/files/9592614/pibooth.log)
[diagnostic.log](https://github.com/pibooth/pibooth/files/9592603/diagnostic.log)

**Environment**
 - Linux - Kubuntu 20.04
 - Microdia Integrated Webcam HD, OpenCV (Dell Laptop), Canon Rebel XS, gPhoto2 (see Secondary bug behavior).
 - Pibooth 2.0.6

Thank you for any help, and thanks for the work on this great project!!!

samuelgbrown, 2022-09-17:
Other interesting behavior:

When I uninstall OpenCV and run pibooth, it runs happily as long as the DSLR (same Canon Rebel XS as above) is plugged in.

With the DSLR connected, if I go through the workflow, it starts the "preview" state and the DSLR camera shutter goes off, but no image ever appears on screen.  I get the "Oops something went wrong" and soft-crash to menu.
[Here is the pibooth.log file.](https://github.com/pibooth/pibooth/files/9592634/pibooth.log)

Side note, looks like the word "crashed" is misspelled in your error message...testament to your coding skills that you haven't heard about that mistake yet ;)

anxuae, 2022-09-18:
Hi @samuelgbrown 

Be sure that the resolution that you have put in the configuration is compatible with your web camera. This OpenCV error seems to tell me that resolution not compatible.

Concerning your **Secondary bug behavior**, this is not a bug but a feature: the `pibooth` **Hybrid mode** which aims at speed up the preview. When you connecte a web cam + DSLR, it take preview with webcam and take capture with DSLR. 

If you uninstall OpenCV, the webcam is no more usable thus preview + capture is done with the DSLR.

Thanks for the mistake report, it seems that nobody read it until now ;-)

samuelgbrown, 2022-09-24:
Sorry about the delay.  Was playing around, and yes: setting the resolution to (640, 480) allows the camera to take a picture, thank you!  Strangely, I could not find a single other resolution that would work, larger or smaller (even (480, 640)), but it works.

The new bug I get is that it captures a single image, but no more.  At the capture state, the webcam light flickers once or twice, the image is captured, then the light turns completely off.  If Pibooth tries to access the camera again, for example for a 4 photo collage ([logs for this scenario here](https://github.com/pibooth/pibooth/files/9639028/pibooth.log)), then it crashes and is unable to restart the camera after its soft-reset.

Do you have any idea how to get the camera to not reset after capture?  A brief look at the configuration options doesn't show any obvious things to try.

Thank you for your help!

anxuae, 2022-09-28:
Hi @samuelgbrown 

This is a strange behavior. By default, pibooth keep in reference the initial resolution as "preview resolution". You have configured a resolution at (640, 480), so after the capture, pibooth comes back to (1280.0, 720.0) which is the initial resolution. However it seems that it is not accepted by the webcam.... strange because it is its initial resolution.

```bash
[ INFO    ] pibooth           : Take a capture
[ DEBUG   ] pibooth           : Taking capture at resolution (640, 480)
[ DEBUG   ] pibooth           : Putting preview resolution back to (1280.0, 720.0)
```

samuelgbrown, 2022-10-10:
I think I'm going to just chalk this up to strange hardware behaviors, probably not worth looking into any more.  Thank you very much for your help regardless!

closed 2022-10-10 by samuelgbrown