pibooth/pibooth #396: Chaining printers

jeyremy, 2022-09-19:
Hi guys, first of all, thanks for this excellent work, i will enter the game of "real photobooth" thanks to you. 

I'm on the way of buying a printer and i was wondering if it's possible to chain printers so we could print on the first one, and when it fails (no paper or no ink) go to the second one this way we could have more prints available... 

i'm wondering if this function + a WCM2Print from DNP would be the killer feature here. 

Maybe there is already a plugin on this ? sorry i'm all new in this. ;

anxuae, 2022-09-21:
Hi @jeyremy 

Printed files are sent to a CUPS server which centralize all the requests. It's difficult to determine if a printer is available or not. Because CUPS does not give this king of info (I guess). Maybe you can have a look to CUPS official documentation if you can find a solution:  http://cups.org

jeyremy, 2022-09-22:
thanks @anxuae for your quick response. 

What i had imagined was get two counters, if i know my first printer has 150 prints max, i could start printing in the second one then.

closed 2022-11-14 by jeyremy