pibooth/pibooth #397: Pibooth with Overlay : Process Stopped

dominiquegrota, 2022-09-21:
Hi,
I'm so glad to use the pibooth for nearly on year now, and it is allways a very fine attraction, and I thank you again for your job !.

When I add a .png file as overlay (with transparency) pibooth crashes while preparing the final picture with a "process stopped" message.
I tried several .png files, several ratio files, several folders to place de .png file.... no hope.

I am also using sometimes a background picture, and this works fine..

Thank's for your help.

Dominique

**Logs**
```
[ DEBUG   ] pibooth           : Plugin found at '/usr/local/lib/python3.7/dist-packages/pibooth/plugins/pibooth_ftp_os.py'
[ INFO    ] pibooth           : Installed plugins: qrcode-1.0.0, picture-template-1.0.0, sound-effects-0.0.3, extra-lights-1.0.1, ftp_os-0.0.1
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ INFO    ] pibooth           : Connected to printer 'Canon_SELPHY_CP1300'
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : took 0.626 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] pibooth           : took 0.337 seconds
[ DEBUG   ] pibooth           : Activate state 'preview'
[ INFO    ] pibooth           : Show preview before next capture
[ DEBUG   ] pibooth           : No translation defined for 'fr/capture' key
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 5653
[ DEBUG   ] pibooth           : Activate state 'capture'
[ INFO    ] pibooth           : Take a capture
[ DEBUG   ] pibooth           : took 0.726 seconds
[ DEBUG   ] pibooth           : Activate state 'processing'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ INFO    ] pibooth           : Saving raw captures
[ DEBUG   ] PIL.TiffImagePlugin: tag: ImageWidth (256) - type: long (4) - value: b'\x00\x00\t\xa0'
[ DEBUG   ] PIL.TiffImagePlugin: tag: ImageLength (257) - type: long (4) - value: b'\x00\x00\x06j'
[ DEBUG   ] PIL.TiffImagePlugin: tag: Make (271) - type: string (2) Tag Location: 46 - Data Location: 134 - value: b'RaspberryPi\x00'
[ DEBUG   ] PIL.TiffImagePlugin: tag: Model (272) - type: string (2) Tag Location: 58 - Data Location: 146 - value: b'RP_imx219\x00'
[ DEBUG   ] PIL.TiffImagePlugin: tag: XResolution (282) - type: rational (5) Tag Location: 70 - Data Location: 156 - value: b'\x00\x00\x00H\x00\x00\x00\x01'
[ DEBUG   ] PIL.TiffImagePlugin: tag: YResolution (283) - type: rational (5) Tag Location: 82 - Data Location: 164 - value: b'\x00\x00\x00H\x00\x00\x00\x01'
[ DEBUG   ] PIL.TiffImagePlugin: tag: ResolutionUnit (296) - type: short (3) - value: b'\x00\x02'
[ DEBUG   ] PIL.TiffImagePlugin: tag: DateTime (306) - type: string (2) Tag Location: 106 - Data Location: 172 - value: b'2022:09:20 18:36:39\x00'
[ DEBUG   ] PIL.TiffImagePlugin: tag: YCbCrPositioning (531) - type: short (3) - value: b'\x00\x01'
[ DEBUG   ] PIL.TiffImagePlugin: tag: ExifIFD (34665) - type: long (4) - value: b'\x00\x00\x00\xc0'
[ DEBUG   ] pibooth           : took 0.482 seconds
[ INFO    ] pibooth           : Creating the final picture
[ INFO    ] pibooth           : Use OpenCvPictureFactory to create background
[ DEBUG   ] pibooth           : took 0.050 seconds
[ INFO    ] pibooth           : Use OpenCvPictureFactory to concatenate images
[ DEBUG   ] pibooth           : took 1.930 seconds
[ INFO    ] pibooth           : Use OpenCvPictureFactory to assemble final image

```
**Environment**
 - OS [Raspbian bullseye]
 - PI 3B+
 - Camera [Rpi V2 camera]
 - qrcode-1.0.0, picture-template-1.0.0, 
 - sound-effects-0.0.3, 
 - extra-lights-1.0.1, 
 - ftp_os-0.0.1 (custom lftp asynchone transfer)

anxuae, 2022-09-21:
Hi @dominiquegrota your log does not show any crash. can you attach the full log?

thanks

dominiquegrota, 2022-09-21:
Hi anxuae, 
it is the full file log !... I tried again and the log file finishes like that. 
Pibooth crashes with a "processus arrêté" in the terminal.

I activated debug mode and take a picture 
[image]

anxuae, 2022-09-28:
Hi @dominiquegrota 

Can you give me your overlay in order to reproduce?

dominiquegrota, 2022-09-28:
Hi @anxuae 
Thank's for your answer, 
here is the overlay I tested. I have tested others before without sucess...

[image]

dominiquegrota, 2022-10-22:
Hi, 
I decided to install a brand new pibooth 2.0.6. And I cannont get overlay working.
here are the logs with the new install, maybe it helps for diagnostic !

Thank-you for your help !
Dominique
```

[ DEBUG   ] pibooth           : Plugin found at '/usr/local/lib/python3.7/dist-packages/pibooth/plugins/pibooth_ftp_os.py'
[ INFO    ] pibooth           : Installed plugins: qrcode-1.0.2, sound-effects-0.0.3, extra-lights-1.0.1, ftp_os-0.0.1
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 3B+
[ DEBUG   ] pibooth           : Fallback to pibooth default camera management system
[ INFO    ] pibooth           : Configuring Picamera camera ...
[ INFO    ] pibooth           : Connected to printer 'Canon_SELPHY_CP1300'
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : BUTTONDOWN: generate CAPTURE button event
[ DEBUG   ] pibooth           : took 1.532 seconds
[ DEBUG   ] pibooth           : Activate state 'choose'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 388
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 441 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 8192
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
[ DEBUG   ] pibooth           : BUTTONDOWN: generate PRINTER event
[ DEBUG   ] pibooth           : took 1.185 seconds
[ DEBUG   ] pibooth           : Activate state 'chosen'
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[... 120 lines cut ...]
  File "/usr/local/lib/python3.7/dist-packages/pibooth/plugins/pibooth_ftp_os.py", line 11, in state_processing_exit
    name = os.path.basename(app.previous_picture_file)
  File "/usr/lib/python3.7/posixpath.py", line 146, in basename
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='raspberrypi', release='5.10.103-v7+', version='#1529 SMP Tue Mar 8 12:21:37 GMT 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.386 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] pibooth           : took 3.352 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 41 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 41 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 59 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 80 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 80 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iTXt' 99 29
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 140 5756
```

werdeil, 2022-11-03:
Pibooth shall work on a Raspberry 3B+ so I don't think it is a performance issue.
Your issue comes from the fact that somewhere the config file you have an empty data while we expect something. Can't tell more with only one line of error.
Check your config, or make a backup and regenerate one

Fotemoucher, 2022-11-11:
Hi, did you solve your overlay problem ? I have the same error, with different types of resolutions / ratio / pictures too

dominiquegrota, 2022-11-11:
No, I did not solve the problem, I tried to reset the original config, but I still get this error...
The pictures I use (png) work as background, but crashes if they are used as overlay.
I have no solution :-(
If someone can give us an advice ? 

Maybe I can add logs in the factory.py. I suppose one variable si not set in line 510, in _build_final_image
    image[y:y+h, x:x+w] = (1.0 - mask) * image[y:y+h, x:x+w] + mask * overlay_image.

Fotemoucher, 2022-11-11:
Yes or can we have an example of a config file and an overlay picture working ?

dominiquegrota, 2022-11-13:
Hi, Maybe @werdeil or @anxuae can give us a working overlay, in cas it comes from our files ?
Thank's

werdeil, 2022-11-13:
@dominiquegrota you can find some overlay to use as examples directly in the repository : https://github.com/pibooth/pibooth/tree/master/tests/captures/portrait

Those worked when we tested, I hope it will help you to find the origin of the issue.

Fotemoucher, 2022-11-13:
Thank you i’ll try too, I use 2 types of capture: 1 or 4 picture, if i properly understand i have to use 2 different overlay like this in the config file:

overlays = (‘/home/user/picture1.png’,´/home/user/picture2.png’)

Is this correct ?

dominiquegrota, 2022-11-14:
Yes ! this is the way to add the 2 picture formats.
I tried the pictures werdeil proposed (thank you !) but i still get an error and no overlay... I will still continue searching...

Fotemoucher, 2022-11-14:
Ok thank both of you, I’ve not tried this overlays yet,
may the error is because of the ratio (camera resolution, margin thick, footers) ?

dominiquegrota, 2022-11-15:
Thank's for these information @Fotemoucher. I tried to set border to 0, remove footers, and set a specific camera resolution (5x bigger than te overlay.
I still have a crash after 
[ INFO    ] pibooth           : Use OpenCvPictureFactory to assemble final image
[ ERROR   ] pibooth           : 
an empty error...

Fotemoucher, 2022-11-15:
I’ll can’t try before this week-end, but if I well understand how it works, (my previous overlays test with my own png), I’ll take a picture with pibooth without overlay, then read its resolution, and then resize one « sure » overlays provided thanks to @werdeil and try it (I have no other idea without the matching config file)

Fotemoucher, 2022-11-18:
Hi, I tried the overlays with my config file, the default config file, tried to play with ratio, tried « apt-get update / apt-get upgrade », nothing worked I have still « precessus stopped », no more idea :(

dominiquegrota, 2022-11-18:
Are you using plugins ? maybe we use a plugin which is failing the overlay ? I use sound-effects, extra-lights, and qrcode

Fotemoucher, 2022-11-18:
Yes I use extra-light, qrcode, and Google photo, but I tried without it, with default config file, same error. Did you tried too?

werdeil, 2022-11-18:
I just tested quickly on a macOS installation:
I added the path to one overlay in a default config file:
overlays = <path to my pibooth>/pibooth/tests/captures/portrait/overlay0.png
Launched pibooth, took a picture and it worked.

I downloaded the overlay provided by @dominiquegrota and it worked too.

It should work whatever the ratio of the overlay or the orientation. The only think that I can't come up is a memory limit on Raspberry Pi that you hit. I'll need to test it again on my pibooth version on a raspberry pi.

Fotemoucher, 2022-11-18:
Thank you for your help, like @dominiquegrota I use pi 3B… I am trying the plug in « pibooth-picture-template » to « bypass » this error but for now i don’t unterstand how it works and have new errors with it :)

dominiquegrota, 2022-12-01:
I tryed again with a RPI3 and a RPI3B+ (I dont' have RPI 4 and don't find any stock !)
I added GPU memory up to 786 MB, without success....

Does anybody get the overlay work with a raspberry 3 ? 

Thank you for your experience feed back...

Fotemoucher, 2022-12-01:
Hi, both yes and no for me: as I didn’t reached to get overlays working, I tried picture template plug in, now i get a new resolution in the final picture rendered. This plug in is very nice, but I don’t reach to set a background with it. To  understand why, once on a test by chance I set in config file an overlays instead of a background, I now have overlays working when I use picture template plug in, and still not working without it (what is weird because the ratio of my overlay png is set to fit to an image ratio matching with final rendered picture without the plug-in, the ratio is not matching with the resolution I now have with the plug in, but it works!)

Fotemoucher, 2022-12-03:
You’re welcome! Can you still use background with picture template plug in ?

dominiquegrota, 2022-12-05:
No I tried, but it does not work : OUPS !
If you need backgound you can unable the plugin. 
I do not know if it works, but you should try to add a picture  in the background of the diagram with value "backgrounds", but you will need to have allways set a png or jpg in the pibooth.cfg for backgrounds ->not (255, 255, 255)

Fotemoucher, 2022-12-05:
Yes but with or without the plug in, the ratio is not the same, so the picture and the resolution change too, a lot modification in config file to deal with it, I will still try to use both background and overlay, and still with picture template i like this plug in !
I read you are using a canon selphy and qr code with a router without Google photo, I Would to the same, is there a way to ask you a few question outsider of this topic if you’re ok ?

dominiquegrota, 2022-12-11:
@anxuae or @werdeil you can close this issue. 
Maybe you could add in the docs that you need to have to use the plugin picture template installed to get the overlay work.

Thank you for the job ! 

Wonderfull app !

werdeil, 2022-12-11:
I won't close it as I think it is not really linked to the plugin but to the picture factory we are using. Overlay is working with OpenCV one but not on the default one. Adding the plugin makes the switch to the opencv one.

So there is a bug on one factory and I'll try to reproduce now that I have an idea.

dominiquegrota, 2023-01-03:
@Fotemoucher : I answered in MP... did you recieve the mail ?