pibooth/pibooth #399: Use Both webacam (or rpi camera) and dslr Camera

jeyremy, 2022-10-06:
Hi again, 

I jusst wanted to ask because i'm not sure ant did not see this in documentation, 
is it possible to have the preview from rpi cam but the final picture taken by DSLR Camera to improve quality ? 

Maybe it's by default like this ? did not see anything mentioning it so i'm not sure... feel free to close this if i misunderstood something

jeyremy, 2022-10-06:
i did find out that's actually how it work !! Amazing. 

Maybe add a mention somewhere in documentation would be great (unless i totally missed it... )

werdeil, 2022-10-10:
It is mentionned in the note of the run pibooth https://pibooth.readthedocs.io/en/latest/sources/start.html?highlight=hybrid#run-pibooth
I guess it is enough?

jeyremy, 2022-10-11:
@werdeil i did not see this... i think i'm used to not read what's in info's that's not a clever move from me. 
thanks for the response ! i'm closing this