pibooth/pibooth #400: Pibooth crashes after each picture, pictures aren't saved

Nirvje, 2022-10-15:
Hello,

Apologies in advance, I'm not an expert.

**Describe the bug**
My raspberry crashes, I do run pibooth through the prompt, it asks me whether I want to take 1 or 4 photos, in both cases, I see one photo taken on screen however the app crashes, the OS crashes, I'm unable to do anything and I must hard reboot the raspberry pi by power it off and on again. The photo taken is not saved anywhere.

**Logs**
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: 
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1824) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Failed to allocate required memory.
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (914) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ INFO    ] pibooth           : Configuring OpenCV camera ...
[ INFO    ] pibooth           : Connected to printer 'Microsoft_Print_to_PDF_via_VNC_de_NIRV'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 captures selected)
[ INFO    ] pibooth           : Show preview before next capture
[ INFO    ] pibooth           : Take a capture
*crash*

**Environment**
I use Raspberry Pi 4 & Raspberry Camera module V2
Pibooth 2.0.6
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian

Thanks in advance for the hand and sorry again as I'm a real beginner

regards,

stef61b, 2022-10-18:
what's parameter in pibooth.cfg
[CAMERA]
# Resolution for camera captures (preview will have same aspect ratio)
# a inverser en fonction de portrait ou paysage
# pour un portrait
#resolution = (1934, 2464)
# pour un paysage
#resolution = (2464,1934)
resolution = (3280,2464) 

and in raspberry configuration how many GPU memory do you have 128Mb ? 256 M

Nirvje, 2022-10-19:
Hello Stef,

Thank you for the reply.

I got the following settings, i'm not sure if i should change them  :

/home/pi/?/pibooth.cfg
[CAMERA]
resolution = (1934, 2464)

_

I have 128MB as GPU memory inside raspi-config / perf options /  GPU memory

Regards,

stef61b, 2022-10-19:
try with 256 Mb . 

Do you have activate camera inside raspi-config ?

Nirvje, 2022-10-19:
Great, it doesn't crash anymore!

Inside raspi-config : 

I have turned the GPU memory  to 256.

I have activated the Interface / Legacy Camera : ON

However, this what the camera sees now : https://ibb.co/wg7BSCD

[img]https://ibb.co/wg7BSCD[/img]

Any clue why ?

werdeil, 2022-10-19:
I see that you are using Bullseye, pibooth is not compatible with the new picamera integration. Can you retry from a Buster install?

Nirvje, 2022-10-19:
Let me try setting a new installation with buster and retry everything :)

Nirvje, 2022-10-19:
tyvm, i'll let you know the outcome

anxuae, 2023-03-02:
New news since several months, please re-open if you are still stuck

closed 2023-03-02 by anxuae