pibooth/pibooth #401: Cannot identify Image File on Mac OS

dpeet, 2022-10-20:
**Describe the bug**
This is basically the same as #311 but that one was closed for inactivity.  I posted there but figured I should make a new issue instead.

When running Pibooth the app successfully takes a picture in 4 or 1 picture mode, but when trying to actually save the capture from the buffer pibooth crashes.  

I downloaded the latest development version of pibooth. I ran
gphoto2 --capture-image-and-download --debug --debug-logfile=capture.log
and was able to successfully capture (and transfer) a picture)

then I ran
pibooth-diag

I put all the files in a google folder since some were large
https://drive.google.com/drive/folders/1O3n9YWYbfY_zEyZXbemNPqicK37PKjGz?usp=sharing

diagnostic.log
diagnostic.raw
capture.log
img_0655.cr3
img_0655.jpg
pibooth.log that kicked this all off

Let me know if there's any other diagnostics to run or something else that would help!

As a note, I also tried it on a Linux box that failed at the same place, so I suspect it's an interaction between the camera and python, and not an OS issue

**Environment**

running pibooth   Python 3.10.8.
 - OS: m1 macbook pro
 - Camera: canon r6
 - Pibooth version 2.0.6 and latest master branch

anxuae, 2023-03-02:
Hello @dpeet 

Can you open a Python shell and type:

```python
>>> from PIL import Image
>>> Image.__version__
```

werdeil, 2024-06-22:
Closing this old issue. Feel free to reopen it if it is still valid.

closed 2024-06-22 by werdeil