pibooth/pibooth #404: HELP: How to set FTP remote path

delmeer, 2022-10-28:
Hi, 

Trying to upload the pictures to an ftp folder on my Synology nas.

Looks like the script is working but how to set the remote path?
Picures need to be uploaded to /photo/pibooth/project1

"""Plugin to upload pictures on a FTP server."""

import os
from ftplib import FTP
import pibooth

__version__ = "0.0.2"

@pibooth.hookimpl
def pibooth_startup(app):
    app.ftp = FTP()
    app.ftp.set_debuglevel(0)
    app.ftp.connect("102.108.108.55", 21)
    app.ftp.login("bla", "blalbla")

@pibooth.hookimpl
def state_processing_exit(app):
    name = os.path.basename(app.previous_picture_file)

    with open(app.previous_picture_file, 'rb') as fp:
        app.ftp.storbinary('STOR {}'.format(name), fp, 1024)

@pibooth.hookimpl
def pibooth_cleanup(app):
    app.ftp.close()

anxuae, 2022-10-31:
Hi, you can have a look to FTP library https://docs.python.org/3/library/ftplib.html

You have the `ftp.cwd` command to change working directory

closed 2022-10-31 by anxuae

delmeer, 2022-10-31:
@anxuae Thanks!!

Shoud be something like:

app.ftp.connect("192.168.178.55", 21)
app.ftp.login("administrator", "1234567890")
app.ftp.cwd("/photo/project1")

Gone give it a try tonight

delmeer, 2022-11-03:
Problem solved. Idea to change it in the default script?

"""Plugin to upload pictures on a FTP server."""

import os
from ftplib import FTP
import pibooth

__version__ = "0.0.3"

@pibooth.hookimpl
def pibooth_startup(app):
    app.ftp = FTP()
    app.ftp.set_debuglevel(0)
    app.ftp.connect("ftp.pibooth.org", 21)
    app.ftp.login("pibooth", "1h!gR4/opK")
    app.ftp.cwd("/path/to/remote/folder")

@pibooth.hookimpl
def state_processing_exit(app):
    name = os.path.basename(app.previous_picture_file)

    with open(app.previous_picture_file, 'rb') as fp:
        app.ftp.storbinary('STOR {}'.format(name), fp, 1024)

@pibooth.hookimpl
def pibooth_cleanup(app):
    app.ftp.close()