pibooth/pibooth #405: Oops! It seems that pibooth has crached

totos21, 2022-11-04:
hello

I have installed pibooth with the documentation on a laptop with the last ubuntu. 

pibooth start but when the photo is taken, I have the error bellow : 

[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (2075) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (651) startPipeline OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (1257) setProperty OpenCV | GStreamer warning: no pipeline
[ ERROR   ] pibooth           : Can not capture frame
[ ERROR   ] pibooth           : system='Linux', node='pibooth', release='5.15.0-52-generic', version='#58-Ubuntu SMP Thu Oct 13 08:03:55 UTC 2022', machine='x86_64', processor='x86_64'
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ WARN:0] global ./modules/videoio/src/cap_gstreamer.cpp (616) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

thank you for your help and sorry for my poor english

facugon85, 2023-01-28:
Uso una camara USB Logitech C270
Instale Gphoto2 y OpenCV  desde https://snapcraft.io/gphoto2 y https://pypi.org/project/opencv-camera/

anxuae, 2023-03-02:
Hi @totos21 

Seems that your camera is not compatible with OpenCV

werdeil, 2024-06-22:
Closing this old question. Feel free to reopen it if it is still valid.

closed 2024-06-22 by werdeil