pibooth/pibooth #406: I can't connect pibooth to my ftp server.

theblackhat17, 2022-11-08:
Hey! 
When I enter my credential informations (host / port / id / password), raspberry can't connect and Pibooth doesn't work..
I don't know if it's 21 or 22 port for ftp. 
I bought domain on IONOS.

_________

Hey!
Lorsque j'entre mes informations d'identification (hôte/port/id/mot de passe), la raspberry n'arrive pas à se connecter et Pibooth ne fonctionne pas..
Je ne sais pas si c'est le port 21 ou 22 pour ftp.
J'ai acheté un domaine sur IONOS.

grenagit, 2022-11-09:
@theblackhat17, We’re gonna need more details so we can help you...

- What is the error message?
- How do you try to connect PiBooth via FTP?
- Did you test your connection to the FTP server without PiBooth?
- At the port level, did you apply the configuration indicated by IONOS?

> In general, port 21 corresponds to [FTP](https://wikipedia.org/wiki/File_Transfer_Protocol) and port 22 to [sFTP](https://wikipedia.org/wiki/SSH_File_Transfer_Protocol)!

*Personally, I wrote a small plugin in PiBooth to send the photos to my server via sFTP. I’ve never had a problem!*

anxuae, 2023-03-02:
Thread relative to FTP server configuration.
I'm closing it since there is no new element and it is out of pibooth scope

closed 2023-03-02 by anxuae