pibooth/pibooth #407: How to configure background photo and icons

Benoit1478, 2022-11-21:
Hello, I am looking on the different sites for how to customize pibooth, in particular the background of the photo, as well as the black background with the icons to change the montage

werdeil, 2022-11-21:
Hi @Benoit1478 , background of the picture is easily configurable through the `backgrounds` key in the config file, in the `[PICTURE]` section. You can either put a RGB tuple or a path to an image. By default you have `backgrounds = (255, 255, 255)` for white.
About the "black background with the icons": You can change the background through the `background` key in the config (in the `[WINDOW]`section. Same you can put either a RGB or a path, default is `background = (0, 0, 0)` for black

We didn't allow the change of the icons, for this you'll have to go in the code to change them (and handle the size + RGBA).

Benoit1478, 2022-11-21:
Bonjour,

Merci pour ta réponse rapide !

ton Pibooth est une merveille , Félicitations !!, et merci pour les informations, je vais de ce pas voir comment effectuer le changement, beaucoup de choses sont bien décrite, mais concernant les changements comme ceux ci, c’est pas très clair.

je viens d’essayer , je suis un touche a tout, mais je ne connais pas vraiment.
j’ai remplacé ( 255,255,255 ) par /home/pi/download/monimage.jpg 

mais après j’ai une erreur.

concernant l’image qui est noir derrière comme sur la photo ci dessous, est -il possible de mettre autre chose ? une image par exemple 

pour les petit bonhomme , si je change les photos qui sont en .png ca pourrais fonctionner ?

werdeil, 2022-11-21:
Let's keep the comments in English so everyone can help you.

You need to put quotes around the path like `'/home/pi/background.jpg'`

About changing the icons as said earlier you need to change them in the code, you can try to put another .png file but no garantee.

Benoit1478, 2022-11-22:
I managed to do everything I want for the moment, is there access to a photo gallery for the reprint? a plugin?

waryeti-us, 2022-11-23:
can someone please provide path to config file. i can't seem to find it after updating. it was originally in ~/.config/pibooth/

Benoit1478, 2022-11-25:
> I managed to do everything I want for the moment, is there access to a photo gallery for the reprint? a plugin?

werdeil, 2022-11-25:
@waryeti-us it shall still be in `~/.config/pibooth`
@Benoit1478 We chose to limit as much as possible the time when pibooth is not available for a new photo, so we did'nt implement any gallery or reprint inside pibooth.

Benoit1478, 2022-11-25:
ok thanks, do you think it is possible to put a system that could display the photos that have been taken from another machine?, to be able to send them back via email for example

waryeti-us, 2022-11-26:
Thanks i installed it as root which is why i couldnt find it.

dominiquegrota, 2023-01-03:
Hi @Benoit1478, 
I use rsync to copy pictures to an other raspberry over wifi. On the second raspberry I installed pipresents (slideshow with auto update of the folder) I added 2 buttons to select witch picture you want to select. 
I did not permet to print the picture, but maybe it is possible to add this last function...

anxuae, 2023-03-02:
Misc thread => I'm closing it, pleas do not hesitate to open a new issue for specific topic

closed 2023-03-02 by anxuae