pibooth/pibooth #408: arcade button doesn't work

totos21, 2022-11-26:
hello
I'm French (sorry for my English) 
I build a Photo Booth with this really cool application (thank you very much for it) but I can't get my arcade buttons to work. 
I plugged in on gpio and gnd and I have configured PIN number on pibooth configuration file in control section. 
When I see the screen 1 photo or 4 photos, buttons doesn't work. 

Can you help me please ? Many thanks

werdeil, 2022-11-26:
If you reach the 1 or 4 photos, it means that one button works at least one?

If it is the case there may be an issue with the de-bouncing mechanism. You can try to change the value of `debounce_delay` in the config.
https://pibooth.readthedocs.io/en/latest/sources/config/config.html?highlight=bounce

Do you have the references for your arcade button? It may require a specific wiring.

dominiquegrota, 2022-11-26:
Hi, 
Maybe you can test if the button is working with command line :
Raspi-gpio get 18
(18 is the gpio port) 
you can try when you press the button and when the button is not pressed so you can see if the button is recognised.

totos21, 2022-11-27:
thank you for your answer
i have tested gpio but no change when i pull up or down the button :

GPIO 13: level=0 fsel=0 func=INPUT

I'm sure my button work because it's an used button for another project.

I don't know why. if you have an idea ? many thanks

tekilaz, 2022-11-27:
Hello.
Its working on my pibooth with my cheap arcade Button.
One on gpio and other one on gnd
Maybe try with anothers gpio ? Or its a problem with your gpio. It could happen

totos21, 2022-11-28:
I have try another button and gpio but nothing works. 
I have tested command : raspi-gpio set 13 pu. The level is 0 when I push the button or 1 if I do nothing. For me the button and gpio work....

totos21, 2022-11-29:
I have find my problem. I confused PIN number and gpio number..... sorry for that.

werdeil, 2022-11-29:
Thanks for the feedback @totos21, I'm closing this issue then.

closed 2022-11-29 by werdeil