pibooth/pibooth #413: EOFError: Ran out of input ??

Benoit1478, 2022-12-08:
there was no more space on the disk, then I got this error
pibooth does not start anymore

pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
[ INFO    ] pibooth           : Installed plugins: extra-lights-1.0.1, picture-template-1.1.0, sound-effects-0.0.3, qrcode-1.0.2, dropbox-0.0.3
[ INFO    ] pibooth           : Starting the photo booth application on Raspberry pi 4B
Traceback (most recent call last):
  File "/home/pi/.local/bin/pibooth", line 8, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.9/site-packages/pibooth/booth.py", line 490, in main
    app = PiApplication(config, plugin_manager)
  File "/home/pi/.local/lib/python3.9/site-packages/pibooth/booth.py", line 127, in __init__
    self.count = Counters(self._config.join_path("counters.pickle"),
  File "/home/pi/.local/lib/python3.9/site-packages/pibooth/counters.py", line 14, in __init__
    self.load()
  File "/home/pi/.local/lib/python3.9/site-packages/pibooth/counters.py", line 54, in load
    self.data.update(pickle.load(fp))
EOFError: Ran out of input

Benoit1478, 2022-12-08:
@werdeil can you help me ?

Benoit1478, 2022-12-08:
@anxuae  can you help me ?

werdeil, 2022-12-09:
I would say that the counters.pickle file is corrupted. You can try to delete it, it shall be next to your config file.

Benoit1478, 2022-12-09:
Yes it worked, do you have an idea of ​​the problem, is it because there was no more space on the memory card?

Many thanks !

anxuae, 2023-03-02:
Yes file was corrupted because of a lack of space in the memory card.
We can do a fix  to avoid loading corrupted file (but counters will be lost)

anxuae, 2023-03-11:
Fixed in https://github.com/pibooth/pibooth/pull/436

PR #436 (2023-03-11): Fix #413: Reinitialize counters if corrupted
Fix #413
commit 006b3244 (2023-03-11): Fix #413: Reinitialize counters if corrupted (#436)

* Reinitialize counters if corrupted

* Pylint ignore docs/conf.py

* Add tests for language

* Fix pylintrc
closed 2023-03-11 by anxuae