pibooth/pibooth #414: Reset Printer Counter via GUI

Ulmerju, 2022-12-11:
The [PRINTER][max_pages] count is working. And the counter can be reset by opening/closing the settings menu.
Since many use the Canon selphy printers, which do not output the paper status, thank you very much for this.

However, it would be very helpful if a message came up in the usual GUI when the counter was reached. A simple "refill paper or ink" message and a confirmation button.

So that every user knows what to do and can continue without special knowledge.

> + Top feature: Possibility to include an image describing the filling process if an image path is specified.

Ulmerju, 2022-12-18:
I wrote a plugin that checks how many prints are left until max_pages. When the print counter is full, a message appears with instructions and an image.

The user must then confirm the refill.

Unfortunately this currently only works in the wait state - can anyone help me get this query to work immediately after pressing the print button before sending it to the printer?

I had the problem, when I display it during printing, this state only shows up briefly and jumps directly to the next step.

`
    """Plugin to display 'Printer Paper State' at screen before 'printing' state."""
    
    import time
    import pygame
    import pibooth
    from pibooth import pictures, fonts

    
    __version__ = "0.0.1"

    text_error = "Papier oder Farbband leer!    Bitte auffüllen"
    text_done = "Erledigt"
    path_image = "Refill_instrictions.png"
    
    @pibooth.hookimpl
    def state_wait_do(win, app , cfg, events):     
    
        if app.count.printed >= cfg.getint('PRINTER', 'max_pages') and cfg.getint('PRINTER', 'max_pages') != -1:
            
            win_rect = win.surface.get_rect()
    
            # Get best font size according to window size
            font = fonts.get_pygame_font(text_error, fonts.CURRENT,
                                        win_rect.width//1.5, win_rect.height//1.5)
    
            # Clear screen
            if isinstance(win.bg_color, (tuple, list)):
                win.surface.fill(win.bg_color)
            else:
                bg_surface = pictures.get_pygame_image(win.bg_color, win_rect.size, crop=True, color=None)
                win.surface.blit(bg_surface, (0, 0))
    
            # Draw the surface at screen
            text_surface = font.render(text_error, True, win.text_color)
            pos = (win_rect.center[0]- win_rect.width//3, win_rect.top + win_rect.height//4)
            win.surface.blit(text_surface, pos)
            
            # Draw Done button
            text_surface = font.render(text_done, True, win.text_color)
            pos = (win_rect.center[0]+ win_rect.width*0.2,win_rect.center[1] + win_rect.height*0.15)
            win.surface.blit(text_surface, pos)
            
            # Draw picture
            size = [win_rect.width//2, win_rect.height//2]
            picture = pictures.get_pygame_image(path_image, size, hflip=False, vflip=False, color= win.text_color)
            x = int(win_rect.left)  # x = int(win_rect.left + win_rect.width * 0.70 - picture.get_rect().width//2)
            y = int(win_rect.bottom - size[1])  # y = int(win_rect.top + win_rect.height * 0.45)
            picture_pos = (x,y)
            win.surface.blit(picture, picture_pos)
    
            # Force screen update and events process
            pygame.display.update()
            pygame.event.pump()
            events = pygame.event.get()
            
            # wait for usre confirmation
            if app.find_capture_event(events) != None or app.find_print_event(events) != None:
                # reset printed images counter
                app.count.printed = 0

`
[image]

AndyKausG, 2023-07-23:
@Ulmerju can u make a plugin out of it? Otherwise can I please have a hint how to implement this?

dominikdoerr, 2023-09-27:
@Ulmerju Thank you very much for providing the code. For me it works just fine with the waiting state, as I would prefer to show the screen on a reboot if needed. Would you mind if I provide your code with some minor tweaks as a plugin for others to install? Would you provide the image you used?

@AndyKausG I have used the code above more or less exactly as posted and added a setup.py file inspired by one of the official plugins: https://github.com/pibooth/pibooth-qrcode/blob/master/setup.py
Afterwards I could install it with `pip3 install /path/to/the/new/plugin/folder`. It was automatically picked up by pibooth, no additional configuration needed.

Ulmerju, 2023-09-27:
@dominikdoerr Thank you very much for creating the plugin. If you have any questions, I'll be happy to help
[image]

dominikdoerr, 2023-10-12:
I have added the plugin code here: https://github.com/dominikdoerr/pibooth-printer-refill-message
Feel free to test it and leave feedback on what could be improved.

For now I have not managed to publish the plugin on PyPI, as I have never done so and have to familiarize myself with the process first. Maybe I'll get around to do so in the next couple weeks.

tonarec, 2023-12-17:
Hi, thanks for the plugin, it was exactly what I was looking for!
I will try this on my current configuration this week and will give you a feedback.

BramWerbrouck, 2024-06-01:
I have a question about this plugin: how can I make that you have to reset the counter through the menu (by pressing the esc-button) and not just by clicking the "done?" text on the screen

when I press the esc-button to enter the settings-menu (on this refill-page pluing) nothing happens...