pibooth/pibooth #415: gstreamer pipline not created

pcmann2004, 2023-01-02:
I have a fresh install of photopi on rpi 3 b+. I have it running without any I/O pins. Also, I connected a Logitech v-u0031. once it counts down to take a photo, an error of "oops something went wrong" below is the last portion of the log. Can someone  please help me.

"
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 441 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 459 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 480 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 480 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 499 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 536 5653
[ DEBUG   ] pibooth           : Add to buffer the image '1370022576'
[ DEBUG   ] pibooth           : Add to buffer the image '1380119000'
[ DEBUG   ] pibooth           : Add to buffer the image '1370022432'
[ DEBUG   ] pibooth           : Add to buffer the image '1380119000'
[ DEBUG   ] pibooth           : Add to buffer the image '1370022648'
[ DEBUG   ] pibooth           : Add to buffer the image '1380119000'
[ DEBUG   ] pibooth           : Add to buffer the image '1370022456'
[ DEBUG   ] pibooth           : Add to buffer the image '1380119000'
[ DEBUG   ] pibooth           : took 4.435 seconds
[ DEBUG   ] pibooth           : Activate state 'capture'
[ INFO    ] pibooth           : Take a capture
[ DEBUG   ] pibooth           : Taking capture at resolution (1934, 2464)
[ ERROR   ] pibooth           : Can not capture frame
[ DEBUG   ] pibooth           : Back to failsafe state due to error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 53, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 103, in state_capture_do
    app.camera.capture(effect)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 220, in capture
    raise IOError("Can not capture frame")
OSError: Can not capture frame
[ DEBUG   ] pibooth           : took 0.619 seconds
[ DEBUG   ] pibooth           : Activate state 'failsafe'
[ ERROR   ] pibooth           : system='Linux', node='PhotoPi', release='5.15.76-v7+', version='#1597 SMP Fri Nov 4 12:13:17 GMT 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ DEBUG   ] pibooth           : took 2.049 seconds
[ DEBUG   ] pibooth           : Activate state 'wait'
"

pcmann2004, 2023-01-02:
update:
mkdir ~/Pictures/pibooth
chmod 777 pibooth/

now i can atleast proceed further and it save the picture.
Now to deal with the choppy refresh rate

Yorams, 2023-01-04:
I also have a fresh install on a RPI 3b+ with a similar Logitec USB Webcam and getting the same error. I did the mkdir and chmod but that did nothing. 

I read somewhere to add "bcm2835-v4l2" to /etc/modules so i tried that.

Then i get some other errors:
```

(python3:1320): GStreamer-CRITICAL **: 23:18:04.675: 
Trying to dispose element videoconvert0, but it is in PLAYING instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.

(python3:1320): GStreamer-CRITICAL **: 23:18:04.675: 
Trying to dispose element pipeline0, but it is in PAUSED instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.

[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (536) startPipeline OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (1085) setProperty OpenCV | GStreamer warning: no pipeline
[ DEBUG   ] pibooth           : |  Taking capture at resolution (1280, 720)
[ ERROR   ] pibooth           : |  Can not capture frame
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pibooth/booth.py", line 402, in main_loop
    self._machine.process(events)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/states.py", line 53, in process
    hook(cfg=self.cfg, app=self.app, win=self.win, events=events)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_hooks.py", line 265, in __call__
    return self._hookexec(self.name, self.get_hookimpls(), kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 80, in _hookexec
    return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 327, in traced_hookexec
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 33, in from_call
    result = func()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_manager.py", line 324, in <lambda>
    lambda: oldcall(hook_name, hook_impls, kwargs, firstresult)
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 60, in _multicall
    return outcome.get_result()
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_result.py", line 60, in get_result
    raise ex[1].with_traceback(ex[2])
  File "/usr/local/lib/python3.9/dist-packages/pluggy/_callers.py", line 39, in _multicall
    res = hook_impl.function(*args)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/plugins/camera_plugin.py", line 103, in state_capture_do
    app.camera.capture(effect)
  File "/usr/local/lib/python3.9/dist-packages/pibooth/camera/opencv.py", line 220, in capture
    raise IOError("Can not capture frame")
OSError: Can not capture frame
[ ERROR   ] pibooth           : |  system='Linux', node='photobooth', release='5.15.76-v7+', version='#1597 SMP Fri Nov 4 12:13:17 GMT 2022', machine='armv7l', processor=''
 ***********************************************************************************
 *                     Oops! It seems that pibooth has crached                     *
 *     You can report an issue on https://github.com/pibooth/pibooth/issues/new    *
 *                       and post the file: /tmp/pibooth.log                       *
 ***********************************************************************************
[ WARN:0] global ../modules/videoio/src/cap_gstreamer.cpp (501) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

(python3:1320): GStreamer-CRITICAL **: 23:18:04.749: 
Trying to dispose element appsink0, but it is in READY instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.
```

facugon85, 2023-01-28:
Uso una camara USB Logitech C270
Instale Gphoto2 y OpenCV desde https://snapcraft.io/gphoto2 y https://pypi.org/project/opencv-camera/

anxuae, 2023-03-02:
Hi @pcmann2004 

Which version of Pi OS did you installed?