pibooth/pibooth #419: Bypass some states for capture without delay

BermudaStudio, 2023-01-24:
Hello,
I would like to bypass the "select the layout" and "preview" states (screens). 

The goal is to shoot portrait with zero delay between button press and DSLR capture.

Solution #1 = a plugin ?
Solution #2 = the possibility to set config option "preview_delay" to value "0" ?

Thanks for your help.
Rémi

werdeil, 2023-01-25:
Hi @BermudaStudio, you can reach almost what you want through the settings, but there is no easy way to put less than 1s of preview delay.

Can you explain more the use case you want ? Because I don't see why waiting 1s is not acceptable.

BermudaStudio, 2023-01-26:
Hi.
I would like to set up unconventionals photobooths with physical countdowns, external sensors, etc.
No live preview needed.
Regards
Rémi

anxuae, 2023-03-02:
Hi @BermudaStudio 

If you defined in the configuration only one possible layout, the "select the layout" screen will not be shown

anxuae, 2023-03-02:
TO definitly hide the preview, there is no other choice than changing the pibooth code to accept preview_delay=0

PR #443 (2023-03-21): Fix #419 : No preview
Close #419
commit 7dc167b8 (2023-03-21): Fix #419 : No preview (#443)

* Change plugins and fix config reset

* Update the counter in camera and view plugin

* app.capture_date now set at first capture

* Remove unwanted files

* Fix doc
closed 2023-03-21 by BermudaStudio