pibooth/pibooth #421: Support Raspberry pi cam 3 on require installation

klemva, 2023-02-01:
Dear All,
I have a question. Every thread, wich i found here in github is about support on Bullseye and Rpi cam 3. But i don't have Bullseye on raspberry, i have installation from this site: https://pibooth.readthedocs.io/en/latest/ sources/install.html and it works fine!
My Hardware is Rpi 4 with 4GB ram and Raspberry pi cam 2.1. But I would like to use Raspberry pi cam 3. And my question is, Will it work with new cam? With autofocus? Or is it possible to use HDR functionality? If I switch the cams?

I try to look for some documents where I can find this information, but I can't find it.

Thank you for your answer!

Václav

werdeil, 2023-02-01:
Hi @klemva, as I don't own a Rpi 3 cam I can't be sure but from what I understand it needs to have the last lib version which is only available on Bullseye.

The support for Bullseye is quite some work and we didn't find time to do it yet.

klemva, 2023-02-02:
Hi @werdeil thank you for reply! Now i understand it:)

closed 2023-02-24 by klemva