pibooth/pibooth #422: Tutorial

Chiv-Lives, 2023-02-04:
I'm not sure if this is the right place to put this, but...
I've been trying for months to get this to work on my Raspberry Pi 4B. Every tutorial I've found has either been wildly unhelpful, been for a different version of Raspberry Pi, were done around 2 years ago with dead or lost repositories, or I get lost in a series of wiki articles, more confused than when I started.

I would love some help for absolute beginners. I've worked very lightly with Raspberry Pi's and have only ever used them as mini-computers to write with, or just mess around with. I've tried my hardest and cannot find a reliable source for installing the PiBooth with an on-board camera (not USB)

https://www.hackster.io/sabat54i/wedding-photo-booth-with-raspberry-pi-1cea3a
This is the original project I worked with. I based almost everything around this project and have gotten most of the materials for it. We're just not sure able the kind of printer we need. Everyone has said a Canon Selphy, but they're either really hard to find, or too expensive.

Any help is appreciated!

anxuae, 2023-03-02:
Hi @Chiv-Lives 

I'm not sure to understand what is your need exactly? 

If you follow carefully the install step, you should be able to install pibooth https://pibooth.readthedocs.io/en/latest/sources/install.html#install . If you have a picamera, it will be recognized automatically.

Almost all printers can do the job. The choice remain up to your considering the speed, the quality and the price of the printer.

Chiv-Lives, 2023-03-14:
Sorry for the late reply.
So, the issue I was having was with the camera. It was the wrong kind, and I eventually got the correct one.
All of the tutorials I had been using were taking me down unhelpful rabbit holes. I did eventually get the program working.