pibooth/pibooth #424: Strip Pictures : print twice

dominiquegrota, 2023-02-12:
Hi, 
I use now for more than one year pibooth, it's Great !
I bought a DS620 for lots of prints (I have a canon selfy...)
I created a specific layout with 3 pictures on a 2x6 inches picture. It works gerat but if I ask 2 prints per page, I get  twice 4"x6" with my strip centered in the 4x6". Is it possible to get twice 2"x6" ? 
Thank-you
Dominique

werdeil, 2023-02-15:
Hi @dominiquegrota, as we don't have this printer I can't help but my best guess would be to tweak CUPS settings, as pibooth is sending the generated image to CUPS without further processing.

anxuae, 2023-03-18:
Hi @dominiquegrota, If you are using the option `pictures_per_page` , the code is done to always make the copy in the opposite orientation than your picture.

So, your picture is Portrait oriented, so copy will be done on landscape way.

With the `pibooth-picture-template`, you can chose the layout that you want and fit your need 

Note that the option `pictures_per_page` will be removed in pibooth v3.0.0. And it will be possible to code your one layout using new hook `pibooth_setup_printer` like defined here: https://github.com/pibooth/pibooth/blob/3.x/docs/sources/plugins/pibooth_custom_printer.py

closed 2023-03-18 by anxuae