pibooth/pibooth #425: Bug with effects

rodhy, 2023-02-13:
**Describe the bug**
Error while I set effects

**Steps to reproduce the bug**
I set catpures effects in pibooth.cfg:
`captures_effects = ('film', 'cartoon', 'washedout', 'film')`

When I take the 4 photos all is ok but if try to makke ather serie i have an "oops its seems that piboth has crashed" message.

I have an error
`Oops! It seems that pibooth has crached `  

If I remove the effects all is ok.

**Environment**
 - OS : Raspbian Buster
 - Camera : Rpi camera
 - Pibooth version 2.0.6
 
[pibooth.log](https://github.com/pibooth/pibooth/files/10726779/pibooth.log)

werdeil, 2023-02-14:
Indeed I think I found the bug the list of effect is only working once. I added a fix it will be on next release.

closed 2023-02-14 by werdeil

rodhy, 2023-02-15:
Ok thanks.
Do you know when it will be released ?

werdeil, 2023-02-22:
@rodhy I updated the code and tested this time. It shall be good.

commit 2881ebe7 (2023-02-14): Fix #425: reset of the count at exit

Reset of the count was missing, leading to the fact that effects list is not working more than one time.
commit 0a4d7d8c (2023-02-22): Reset count at the processing state

Fix #425: Previous fix was made at the wrong state (as we are exiting capture state for each capture), I moved it to the processing state.