pibooth/pibooth #426: Pybooth does not recognize gPhoto2

correonuriayalejandro, 2023-02-14:
Hello, good evening,

First of all, I wanted to thank you for the work and effort you put into carrying out this application.

I have installed pybooth on my raspberry pi 4 and I am not getting it to work, I attach the LOG below.

A while ago I had it installed and working, I wanted to make a clean installation and I don't understand why it's not working now.

**Logs**
nuriayalejandro@raspberrypi:~ $ gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Canon EOS 700D                 usb:001,012
nuriayalejandro@raspberrypi:~ $ pibooth-diag
pygame 2.1.2 (SDL 2.0.16, Python 3.9.2)
Hello from the pygame community. https://www.pygame.org/contribute.html
pygame-menu 4.0.7
Pibooth version installed: 2.0.6
Installed plugins:
gPhoto2 not installed, cannot diagnose connected DSLR

**Environment**
 - OS Raspbery PI OS Buyelle
 - Camera Canon EOS 700D
 - Pibooth version 2.06

correonuriayalejandro, 2023-02-14:
I installed the 32-bit version of raspbian bulleye and it worked on the first try.

anxuae, 2023-03-02:
Hi @correonuriayalejandro 

Did you installed pibooth with DSLR dependencies? 

`sudo pip3 install -U pibooth[dslr]`

anxuae, 2023-03-18:
Hi @correonuriayalejandro , do you have some news about this issue?

nsty, 2023-09-22:
I had exactly the same problem. Raspberry Pi OS 64-bit (Bullseye, fresh install, up-to-date) and followed the [manual installation procedure](https://pibooth.readthedocs.io/en/latest/sources/install.html#manual-procedure) with the `[dslr,printer]` option.

`gphoto2 --auto-detect` recognizes the camera successfully, not so Pibooth. With a bit of debugging I found that Pibooth drops out already when checking for the gPhoto2 installation in `gphoto.py`. https://github.com/pibooth/pibooth/blob/1229726f248405aa3b28dd12018d5c0ce9e62e2a/pibooth/camera/gphoto.py#L26

As suggested in https://github.com/pibooth/pibooth/issues/387#issuecomment-1235619054, I tried `python -m gphoto2` with an error coming up. So python does not recognize gPhoto at all.

`gphoto2 --version` returns a lower version for the library than the frontend
```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

I found a [similar issue](https://github.com/gphoto/gphoto2/issues/25) in the gPhoto2 repo itself with the "solution"/workaround.

`sudo apt remove libgphoto2-port12`

Solves the issue for me of apparently two conflicting installed versions. Now `gphoto2 --version` returns more recent versions used and python is able to find the library as well is Pibooth.

```I had exactly the same problem. Raspberry Pi OS 64-bit (Bullseye, fresh install, up-to-date) and followed the [manual installation procedure](https://pibooth.readthedocs.io/en/latest/sources/install.html#manual-procedure) with the `[dslr,printer]` option.

`gphoto2 --auto-detect` recognizes the camera successfully, not so Pibooth. With a bit of debugging I found that Pibooth drops out already when checking for the gPhoto2 installation in `gphoto.py`. https://github.com/pibooth/pibooth/blob/1229726f248405aa3b28dd12018d5c0ce9e62e2a/pibooth/camera/gphoto.py#L26

As suggested in https://github.com/pibooth/pibooth/issues/387#issuecomment-1235619054, I tried `python -m gphoto2` with an error coming up. So python does not recognize gPhoto at all.

`gphoto2 --version` returns a lower version for the library than the frontend
```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

I found a [similar issue](https://github.com/gphoto/gphoto2/issues/25) in the gPhoto2 repo itself with the "solution"/workaround.

`sudo apt remove libgphoto2-port12`

Solves the issue for me of apparently two conflicting installed versions. Now `gphoto2 --version` returns more recent versions used and python is able to find the library as well is Pibooth.

```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.31         standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

Not sure how to solve this properly though as it will come up from time to time I guess when versions start to mismatch. I'm also not sure if it's really a problem with the 64-bit version of the OS or if it just comes up there by chance now.
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.31         standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

Not sure how to solve this properly though as it will come up from time to time, I guess, when versions start to mismatch. I'm also not sure if it's really a problem with the 64-bit version of the OS or if it just comes up there by chance now.

nsty, 2023-09-22:
I had exactly the same problem. Raspberry Pi OS 64-bit (Bullseye, fresh install, up-to-date) and followed the [manual installation procedure](https://pibooth.readthedocs.io/en/latest/sources/install.html#manual-procedure) with the `[dslr,printer]` option.

`gphoto2 --auto-detect` recognizes the camera successfully, not so Pibooth. With a bit of debugging I found that Pibooth drops out already when checking for the gPhoto2 installation in `gphoto.py`. https://github.com/pibooth/pibooth/blob/1229726f248405aa3b28dd12018d5c0ce9e62e2a/pibooth/camera/gphoto.py#L26

As suggested in https://github.com/pibooth/pibooth/issues/387#issuecomment-1235619054, I tried `python -m gphoto2` with an error coming up. So python does not recognize gPhoto at all.

`gphoto2 --version` returns a lower version for the library than the frontend.
```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

I found a [similar issue](https://github.com/gphoto/gphoto2/issues/25) in the gPhoto2 repo itself with the "solution"/workaround.

`sudo apt remove libgphoto2-port12`

Solves the issue for me of apparently two conflicting installed versions. Now `gphoto2 --version` returns more recent versions used and python is able to find the library as well is Pibooth.

```
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.31         standard camlibs (SKIPPING docupen lumix), gcc, no ltdl, EXIF
libgphoto2_port 0.12.2         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

Not sure how to solve this properly though as it will come up from time to time, I guess, when versions start to mismatch. I'm also not sure if it's really a problem with the 64-bit version of the OS or if it just comes up there by chance now.