pibooth/pibooth #427: Change GUI interface

jacare0000, 2023-02-15:
Dear all, 

Thanks for this amazing project! 
I am trying some improvements that comunity May Can help. 

So the ideia is change the GUI interface, like the place where the text are show, the place the touch should be reactive to each function etc... 
I may can program this, but i Cant find where is the Code related to this functions. 
Someone Can Point me the .py file where i should work on? 

Thank you!

werdeil, 2023-02-15:
Hi @jacare0000,

About the touch part, the main idea to make pibooth work with touchscreen and GPIO buttons was to associate each action to one of the event that can be triggered by the 2 buttons, you can see the function `find_*_event` in `booth.py`.
For example the forget/print choice is using the capture (left side of screen)/ print (right side of screen) events.

About the texts, you'll find everything in the `window.py`.

anxuae, 2023-03-02:
In v3.0.0, improvement will be done by adding a new hook giving the possibility to redefine the "scene" by custom one

anxuae, 2023-03-17:
New hook `pibooth_setup_states` added to add or customize user interface 
Forecasted in v3.0.0, coming soon

PR #437 (2023-03-15): Fix #431 & #427 : New hooks for printer and scene
Close #431
Close #427
closed 2023-03-17 by anxuae