pibooth/pibooth #428: Button missmatch

flammeur, 2023-02-15:
Hello folks,

I use raspberry 3 + tacticle screen and DLSR camera

Everything was working fine but for some reason the button area to click  (or press on the screen ) doesn't work accuretly anymore.

If i press "forget this picture" it will send it anyways to the printer , i have to click on the very top of the arrow next to the text to work.
and if i click "print it" won't print

I tried to reset conf , reinstall the sd card , put full screen , remove the tacticle screen , tried remotly (vnc) 
Even if i don't install the printer it does it .

I don't get it before if i didn't connect a printer i wouldn't even have the options any idea please ?

jacare0000, 2023-02-15:
Hi Flammeur,

I think the GUI interface is confuse to understand.
You should Click on one side of the screen to forget the photo (left side If I am not wrong) and the other side of the screen to print. Because both text are in the same side of the screen People usually Click on the text and it does the wrong action.
Please try Click left side of the screen and then right side and see If this work.

I already porpose an ehancement for change the GUI in order to make it more easy to use.

Please correct me If I am wrong.
Best regards.

flammeur, 2023-02-15:
Hello,

Thanks for your answer , 

Yeah it's working with left side and right side thanks but i have been using it for over a week and i never got this issue before
the text was reponsive...

Well thanks but yeah not really interesting to use touchscreen

werdeil, 2023-02-15:
Hi @flammeur: as said by @jacare0000, Pibooth is seeing the touchscreen as if there was only two surfaces (left and right) so to forget the photo you need to click on it (or anywhere on the left half of the screen).
You can change the text linked to the forget to make it more explicit. as we are compatible with physical button we cannot put "click on the picture" in the text.

flammeur, 2023-02-22:
Hello @werdeil 
Thanks for your reply 
I get your point I have tested with explicit text but people will still click on the text anyway.

The best way I guess would be to have text on the left side and right side .

Thanks for the project !

werdeil, 2023-02-22:
The issue with the text on the left side is that you either reduce the size of the image displayed (to put text above or below) or you put the text on top of the image but depending on the layout it can hide people or text from it.
As the issue is only on the forget mechanism I think I'll made it optional through the config (as it is not a key feature).

anxuae, 2023-03-02:
Hi, user experience will be inproved ion v3.0.0 as any graphical element at screen will be "clickable""

anxuae, 2023-03-11:
Fixed in 3.0.0 that will be release soon

closed 2023-03-11 by anxuae