pibooth/pibooth #432: Add button mappings in the configuration file

anxuae, 2023-03-02:
Possibly add possibility to change the button mapping in the configuration

From @philipbarwi in #42

orangarot, 2023-04-08:
I would also like a option of setting the two hardware buttons to two different Keys(or mouse left and right click) so i can use it on my Linux machine without gpio pins.

anxuae, 2023-04-08:
Hi @orangarot 

You can already do it: https://pibooth.readthedocs.io/en/latest/sources/start.html#controls

P -> trigger capture event
Ctrl + E -> trigger print event

orangarot, 2023-04-08:
Yes, but i can´t choose a second layout without the need for a third button. With the hardware buttons i can navigate with just two buttons. 
If i want to take two pictures and print them i have to press first P then Arrow Left or Right then to print Ctrl + E, i not sure whah i have to press to delete.

It would be nice if Button One could be mapped to Left arrow key and Button two to Right arrow key. It would making a custom USB Controller much easier. 

Great Software by the way, it runs flawless on an old mac mini with Linux Mint.

anxuae, 2023-04-08:
It will be possible with pibooth v3.0.0 coming soon: https://pibooth.readthedocs.io/en/3.x/sources/start.html#controls

`C` (capture)  and `P` (print) key will simulate hardware button

orangarot, 2023-04-11:
Thank you for your help!